zephyrproject-rtos/zephyr #12825: SystemView Feature Not working

bfrog, 2019-01-29:
Enabling systemview currently does not work as expected, systemview can't find the RTT segment and fails to record.

nashif, 2019-01-29:
@pkral78 I bisected the tree and fount this commit to be the likely offender: 2c84fd2631dcc8e586a8455b7fe696d58a92aaf5

Can you please take a look and see of systemview is working for you?

pkral78, 2019-01-29:
Similar issue here https://github.com/zephyrproject-rtos/zephyr/issues/11393

pkral78, 2019-01-29:
My recommendation was to use another RTT channel, although it may not be supperted by SystemView yet.

We have to reconsider what to do next.

pkral78, 2019-01-30:
I just looked into ext/debug/segger/systemview and found that SystemView is nicely configurable, there some are interesting macros:
`SEGGER_SYSVIEW_LOCK()`
`SEGGER_SYSVIEW_UNLOCK()`
`SEGGER_SYSVIEW_RTT_CHANNEL`

which all defaults to conflicting values - they should be configurable from Kconfig.

Also it is necessary to ensure that `SYS_INIT(rtt_init,...)` in [SEGGER_RTT_zephyr.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/ext/debug/segger/rtt/SEGGER_RTT_zephyr.c) is initialized before `SYS_INIT(sysview_init,...)` in [sysview.c](https://github.com/zephyrproject-rtos/zephyr/blob/master/subsys/debug/tracing/sysview.c)

galak, 2019-03-28:
@nordic-krch Can you take a look at this issue.

mbolivar, 2019-04-23:
With this patch (a revert of the relevant pieces of https://github.com/zephyrproject-rtos/zephyr/commit/2c84fd2631dcc8e586a8455b7fe696d58a92aaf5 that @nashif bisected to), I am able to get SystemView tracing working on an nRF52840.

```
From 320b2428cb87dad467a0b6d83d23c374a00e611e Mon Sep 17 00:00:00 2001
From: Marti Bolivar <marti.bolivar@nordicsemi.no>
Date: Thu, 18 Apr 2019 12:54:25 -0600
Subject: [PATCH] [temphack] SEGGER_RTT_CONF.h: remove zephyr mutex locking

---
 ext/debug/segger/rtt/SEGGER_RTT_Conf.h | 7 +------
 1 file changed, 1 insertion(+), 6 deletions(-)

diff --git a/ext/debug/segger/rtt/SEGGER_RTT_Conf.h b/ext/debug/segger/rtt/SEGGER_RTT_Conf.h
index 142949ad9e..a960ea7439 100755
--- a/ext/debug/segger/rtt/SEGGER_RTT_Conf.h
+++ b/ext/debug/segger/rtt/SEGGER_RTT_Conf.h
@@ -138,12 +138,7 @@ Revision: $Rev: 9599 $
 *       Rowley CrossStudio and GCC
 */
 #if (defined __SES_ARM) || (defined __CROSSWORKS_ARM) || (defined __GNUC__)
-  #ifdef __ZEPHYR__
-    #include <kernel.h>
-    extern struct k_mutex rtt_term_mutex;
-    #define SEGGER_RTT_LOCK() k_mutex_lock(&rtt_term_mutex, K_FOREVER);
-    #define SEGGER_RTT_UNLOCK() k_mutex_unlock(&rtt_term_mutex);
-  #elif __ARM_ARCH_6M__
+  #ifdef __ARM_ARCH_6M__
     #define SEGGER_RTT_LOCK()   {                                                                   \
                                     unsigned int LockState;                                         \
                                   __asm volatile ("mrs   %0, primask  \n\t"                         \
-- 
2.21.0

```

pkral78, 2019-04-24:
Oh, what you try to do ? I think we are still going in circles. Please read discussion in  #11393.

mbolivar, 2019-04-24:
> Oh, what you try to do ? I think we are still going in circles. Please read discussion in #11393.

Use SystemView to trace execution, e.g. by enabling it on the philosophers sample. I've read that discussion.

pkral78, 2019-04-25:
I would prefer configure SystemView to use differend RTT channel if it is possible, it is cleanest solution. A smarter RTT ring buffer locking as suggested in #11393 is on my TODO list. (OT: unfortunately I'm currently stuck on issues that both HEADs of bluez mesh stack and zephyr are still incompatible).

ioannisg, 2019-06-27:
@nordic-krch could you give some update on this issue?

nordic-krch, 2019-07-01:
@ioannisg i haven't been looking at it. I've been assigned but frankly speaking i haven't been using systemview at all and there is planty on my plate now so i don't think that i will be looking at it in next few weeks.

nashif, 2019-08-29:
can't reproduce, works for me with `samples/synchronization/` and the following changes:

```patch
diff --git a/samples/synchronization/prj.conf b/samples/synchronization/prj.conf
index fb9f4cdb23..7a1792ae40 100644
--- a/samples/synchronization/prj.conf
+++ b/samples/synchronization/prj.conf
@@ -1,3 +1,5 @@
 CONFIG_STDOUT_CONSOLE=y
 # enable to use thread names
-#CONFIG_THREAD_NAME=y
+CONFIG_THREAD_NAME=y
+CONFIG_SEGGER_SYSTEMVIEW=y
+CONFIG_USE_SEGGER_RTT=y
```

closed 2019-08-29 by nashif