zephyrproject-rtos/zephyr #12827: nrf: nrf_power_gpregret2_get: u32_t or u8_t

chris-makaio, 2019-01-29:
Dear All,

I'm not really sure if this CAN be enhanced, but I wonder why nrf_power_gpregret2_set and nrf_power_gpregret2_get handle a uint8_t - shouldn't that be u32_t?
At least in the doc it seems it has 4 bytes

anangl, 2019-02-01:
Each of these registers can hold only an 8-bit value. The registers occupy 32-bit locations but only lowest 8 bits in each is implemented. See for instance [the documentation for nRF52840](https://www.nordicsemi.com/DocLib/Content/Product_Spec/nRF52840/latest/power?166#register.GPREGRET) (row "ID" in the table, only bits marked with `A` are implemented).

> At least in the doc it seems it has 4 bytes

Did you find any other information suggesting that these registers can hold 32-bit values?

chris-makaio, 2019-02-01:
Understood.
Well, I‘ve seen a few posts (by users) in DevZone which are stating that they‘re 32bit.
However, can be closed :)

closed 2019-02-01 by chris-makaio