zephyrproject-rtos/zephyr #12844: ARC MPU version 3 is configured incorrectly

andrewboie, 2019-01-29:
Unlike all our other targets, the ARC MPU version 3 is configured so that supervisor mode does not have default access to RAM.

This was covered up by a hack in the app shared memory test, which had to disable the MPU in order to initialize application shared memory bss:

```
#if defined(CONFIG_ARC)
	/*
	 * appmem_init_app_memory will access all partitions
	 * For CONFIG_ARC_MPU_VER == 3, these partitions are not added
	 * into MPU now, so need to disable mpu first to do app_bss_zero()
	 */
	arc_core_mpu_disable();
	appmem_init_app_memory();
	arc_core_mpu_enable();
#else
	appmem_init_app_memory();
#endif
```

This is not correct. Supervisor mode (the kernel) needs free access to the entirety of system RAM, we are only using memory domains to control user mode access to memory.

Please fix the ARC MPU ver 3 background map to allow supervisor mode access, so that the policy is the same for ARC MPU version 2, ARM, and x86.

vonhust, 2019-01-30:
one proof  about we can't enable the background region is that in tests/kernel/mem_protect/syscall/src/main.c, line 128

ret = string_nlen((char *)0xFFFFFFF0, BUF_SIZE, &err);
zassert_equal(err, -1, "nonsense string address did not fault");

if we enable the background region to let kernel mode access all memory map,  this cannot be caught by MPU. so the test will fail.

The root cause of 

	arc_core_mpu_disable();
	appmem_init_app_memory();
	arc_core_mpu_enable();

is ARC MPU v3 doest not support MPU overlap, not background region.

andrewboie, 2019-01-30:
> Sorry, I'm still not convinced opening up all RAM for privileged threads is a good thing. Principle of least privilege: why opening up if not needed?

We discussed this already. There is no design for restricting access to system RAM to the kernel itself. If the kernel faults when trying to zero BSS that is fundamentally broken. The kernel will not be able to access kernel objects, when Adithya was enabling app shared memory by default to all tests, he was having to do crazy stuff like moving kernel objects into application memory. The whole thing goes up in flames. This was never intended in the current design.

Like I said before, if we want to introduce the concept of restricting access to system RAM, we can discuss it, come up with a proposal, etc. I'm not convinced it adds any security value since supervisor mode can reprogram the MPU at will, but it could add some assistance to debugging I guess.

But right now, ARC is acting fundamentally differently that every other arch that supports this stuff. We can't have this situation.

> one proof about we can't enable the background region is that in tests/kernel/mem_protect/syscall/src/main.c, line 128
>
> ret = string_nlen((char *)0xFFFFFFF0, BUF_SIZE, &err);
> zassert_equal(err, -1, [...]

It doesn't need to be caught by the MPU.
If you try to access out-of-bounds RAM, you get some other exception right? Trap that instead. The point of that mechanism is to test looking at that memory range in a controlled condition such that if we get any kind of fault, the kernel will gracefully return an error instead of exploding. It does not have to specifically be an MPU fault.

andrewboie, 2019-01-31:
> I'll think of raising a PR/issue for a more constrained privileged mode access model - after LTS.

OK cool.

Off the top of my head:

- Need to understand how kernel memory will be partitioned up
- Need an access policy for the system workqueue thread
- Need an access policy for the kernel heap, other common shared data structures
- Need a policy for when interrupts happen. They can't just use the domain of the thread that was interrupted.
- Need a policy for system calls. Many system call will need to access kernel memory, and still have access to the user thread caller's memory for buffer data copying and so forth

ruuddw, 2019-01-31:
Yes, interrupts handlers requiring access would be tricky. Others have a thread context that could be explicitly granted access using the memory domain api.

PR #12906 (2019-02-06): add the handling of APP_SHARED_MEM in arc mpu driver

* add the handling of APP_SHARED_MEM. privileged threads can access all the mem
explictly defined in user mode, i.e., APP_MEM & APP_SHARED_MEM
* remove arc specific related code in tests/kernel/mem_protect/userspace. no need to disbale mpu to do app mem init because of changes in ARC MPU driver

Fixes #12844
closed 2019-02-06 by nashif