zephyrproject-rtos/zephyr #12860: No test builds these files

galak, 2019-01-30:
arch/arc/core/atomic.S
arch/arm/core/cortex_m/cmse/arm_core_cmse.c
arch/arm/core/cortex_m/tz/arm_core_tz.c
arch/x86_64/core/demo-kernel.c
arch/x86_64/core/qemuinc.c
arch/x86_64/core/xuk-stub16.c
arch/x86_64/core/xuk-stub32.c
arch/x86_64/core/xuk-stubs.c
arch/x86/core/reboot_rst_cnt.c
arch/x86/core/spec_ctrl.c
arch/xtensa/core/irq_manage.c
arch/xtensa/core/swap.S
arch/xtensa/core/thread.c
arch/xtensa/core/xtensa_context.S
arch/xtensa/core/xtensa_intr_asm.S
arch/xtensa/core/xtensa_intr.c
arch/xtensa/core/xtensa_vectors.S
arch/xtensa/core/xt_zephyr.S
boards/arc/iotdk/arc_mpu_regions.c
boards/arm/v2m_musca/pinmux.c
boards/posix/native_posix/sdl_events.c
boards/posix/nrf52_bsim/argparse.c
boards/posix/nrf52_bsim/bstests_entry.c
boards/posix/nrf52_bsim/irq_handler.c
boards/posix/nrf52_bsim/k_busy_wait.c
boards/posix/nrf52_bsim/main.c
boards/posix/nrf52_bsim/time_machine.c
boards/posix/nrf52_bsim/trace_hook.c
boards/riscv32/rv32m1_vega/pinmux.c
boards/xtensa/intel_s1000_crb/pinmux.c
drivers/aio/aio_comparator_handlers.c
drivers/audio/decimation/pdm_decim_int32_02_4288_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_02_4375_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_03_3850_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_03_4375_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_04_4375_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_05_4331_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_06_4156_5100_010_095.c
drivers/audio/decimation/pdm_decim_int32_08_4156_5380_010_090.c
drivers/audio/decimation/pdm_decim_table.c
drivers/audio/intel_dmic.c
drivers/audio/tlv320dac310x.c
drivers/can/can_handlers.c
drivers/clock_control/clock_control_rv32m1_pcc.c
drivers/console/ram_console.c
drivers/counter/counter_dtmr_cmsdk_apb.c
drivers/counter/counter_handlers.c
drivers/counter/counter_tmr_cmsdk_apb.c
drivers/counter/timer_dtmr_cmsdk_apb.c
drivers/counter/timer_tmr_cmsdk_apb.c
drivers/crypto/crypto_ataes132a.c
drivers/display/display_dummy.c
drivers/display/display_sdl.c
drivers/dma/dma_cavs.c
drivers/dma/dma_handlers.c
drivers/dma/dma_sam_xdmac.c
drivers/entropy/entropy_esp32.c
drivers/ethernet/eth_stellaris.c
drivers/flash/flash_gecko.c
drivers/flash/flash_sam0.c
drivers/flash/flash_sam.c
drivers/flash/flash_stm32f0x.c
drivers/flash/flash_stm32f3x.c
drivers/flash/flash_stm32f3x_priv.c
drivers/flash/flash_stm32f7x.c
drivers/flash/soc_flash_qmsi.c
drivers/flash/spi_nor.c
drivers/gpio/gpio_atmel_sam3.c
drivers/gpio/gpio_esp32.c
drivers/gpio/gpio_rv32m1.c
drivers/i2c/i2c_esp32.c
drivers/i2c/i2c_gpio.c
drivers/i2c/i2c_sam_twi.c
drivers/i2s/i2s_cavs.c
drivers/i2s/i2s_handlers.c
drivers/i2s/i2s_sam_ssc.c
drivers/ieee802154/ieee802154_cc1200.c
drivers/interrupt_controller/cavs_ictl.c
drivers/interrupt_controller/dw_ictl.c
drivers/interrupt_controller/loapic_spurious.S
drivers/interrupt_controller/rv32m1_intmux.c
drivers/ipm/ipm_handlers.c
drivers/led/led_handlers.c
drivers/led_strip/apa102.c
drivers/neural_net/intel_gna.c
drivers/pci/pci_shell.c
drivers/pinmux/dev/pinmux_dev_arm_beetle.c
drivers/pinmux/dev/pinmux_dev_stm32.c
drivers/pinmux/dummy.c
drivers/pinmux/pinmux_esp32.c
drivers/pinmux/pinmux_intel_s1000.c
drivers/pinmux/pinmux_rv32m1.c
drivers/pwm/pwm_dw.c
drivers/pwm/pwm_imx.c
drivers/pwm/pwm_led_esp32.c
drivers/pwm/pwm_nrfx.c
drivers/sensor/pms7003/pms7003.c
drivers/sensor/qdec_nrfx/qdec_nrfx.c
drivers/serial/uart_esp32.c
drivers/serial/uart_pl011.c
drivers/serial/uart_rv32m1_lpuart.c
drivers/timer/rv32m1_lptmr_timer.c
drivers/timer/sam0_rtc_timer.c
drivers/watchdog/iwdg_stm32.c
drivers/watchdog/wdt_esp32.c
drivers/watchdog/wdt_nrfx.c
kernel/int_latency_bench.c
lib/gui/lvgl/lvgl_color_16.c
lib/gui/lvgl/lvgl_color_8.c
lib/gui/lvgl/lvgl_fs.c
lib/gui/lvgl/lvgl_mem_kernel.c
lib/gui/lvgl/lvgl_mem_user.c
soc/arm/arm/musca_a/arm_mpu_regions.c
soc/arm/arm/musca_a/soc.c
soc/arm/nordic_nrf/nrf91/mpu_regions.c
soc/arm/nordic_nrf/nrf91/soc.c
soc/riscv32/openisa_rv32m1/soc.c
soc/riscv32/openisa_rv32m1/soc_irq.S
soc/riscv32/openisa_rv32m1/vector.S
soc/riscv32/openisa_rv32m1/wdog.S
soc/xtensa/esp32/esp32-mp.c
soc/xtensa/esp32/soc.c
soc/xtensa/intel_s1000/soc.c
subsys/bluetooth/controller/ll_sw/ll_mesh.c
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_prof.c
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_test.c
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_tmp.c
subsys/bluetooth/controller/ll_sw/ull_tmp.c
subsys/debug/tracing/cpu_stats.c
subsys/disk/disk_access_sdhc.c
subsys/logging/log_backend_swo.c
subsys/logging/sys_log.c
subsys/logging/sys_log_net.c
subsys/net/l2/ieee802154/ieee802154_radio_aloha.c
subsys/net/l2/openthread/openthread.c
subsys/net/l2/openthread/openthread_utils.c
subsys/net/lib/openthread/platform/alarm.c
subsys/net/lib/openthread/platform/diag.c
subsys/net/lib/openthread/platform/flash.c
subsys/net/lib/openthread/platform/logging.c
subsys/net/lib/openthread/platform/misc.c
subsys/net/lib/openthread/platform/platform.c
subsys/net/lib/openthread/platform/radio.c
subsys/net/lib/openthread/platform/random.c
subsys/net/lib/openthread/platform/shell.c
subsys/net/lib/openthread/platform/spi.c
subsys/power/pm_ctrl.c
subsys/power/policy/policy_dummy.c
subsys/shell/shell_rtt.c

aescolar, 2019-01-30:
boards/posix/nrf52_bsim/* are covered if you BabbleSim is found when running sanitycheck

nashif, 2019-01-30:
Building is one thing, testing is a another, this gives an idea on our test coverage:

https://builds.zephyrproject.org/coverage/20190128/index.html

using qemu_x86 and mps2_an385

galak, 2019-01-30:
> Building is one thing, testing is a another, this gives an idea on our test coverage:
> 
> [builds.zephyrproject.org/coverage/20190128/index.html](https://builds.zephyrproject.org/coverage/20190128/index.html)
> 
> using qemu_x86 and mps2_an385

Agreed, my first concern was just the lack of even a build test for some code, or if some code was dead.

carlescufi, 2019-01-30:
In the nrf side:
```
drivers/pwm/pwm_nrfx.c
drivers/sensor/qdec_nrfx/qdec_nrfx.c
drivers/watchdog/wdt_nrfx.c
```
Those need to be built and I see no reason not to. @anangl can you please add those in the relevant sections?
```
soc/arm/nordic_nrf/nrf91/mpu_regions.c
soc/arm/nordic_nrf/nrf91/soc.c
```
I am surprised that those are not ever built. Any ideas @ioannisg?

```
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_prof.c
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_test.c
subsys/bluetooth/controller/ll_sw/nordic/lll/lll_tmp.c
subsys/bluetooth/controller/ll_sw/ull_tmp.c
```
Those are fine they are profiling and test/tmp files. I don't think we need to build them at all by default. Agree @cvinayak ?

galak, 2019-01-30:
> I am surprised that those are not ever built. Any ideas @ioannisg?

I think that's because the toolchain I'm using is the older 0.9.x so arm-v8m stuff doesn't build/

ioannisg, 2019-01-30:
`arm_core_cmse.c `is always built when ARM ARMv8-M is used.

```
add_subdirectory_ifdef(CONFIG_CPU_CORTEX_M_HAS_CMSE cortex_m/cmse)
add_subdirectory_ifdef(CONFIG_ARM_SECURE_FIRMWARE cortex_m/tz)
```
`tz.c` requires `CONFIG_TRUSTED_EXECUTION_SECURE`

cvinayak, 2019-01-30:
@carlescufi assigned myself, I will include the controller files under tests so as to build them.

vanwinkeljan, 2019-01-31:
Regarding the SDL based display driver (drivers/display/display_sdl.c) this would require a 32-bit SDL library in the CI docker container including all its dependencies (X11, ...), if this is not an issue the samples can easily be added to CI.

cvinayak, 2019-02-04:
@galak How could I (or you help) verify that I have handle the missing building of the files in #13022 ? Please approve it subsequently.

aescolar, 2019-02-04:
@cvinayak : For your case, you should be able to do something like this:
```
sanitycheck  -T tests/bluetooth/init/  -v
find sanity-out -name "*obj*" |  cut -d / -f 8-  | sort | uniq | less
```
If you see the files in that list of objects they are built.

erwango, 2019-02-12:
I'm looking at STM32 files. It seems that remaining files would be:
```
drivers/flash/flash_stm32f0x.c
drivers/flash/flash_stm32f3x.c
drivers/flash/flash_stm32f3x_priv.c
drivers/flash/flash_stm32f7x.c
```
Since following are not reported, I assume L4/F4 STM32 boards are built in some sample/application, need to add missing SoCs to be tested on it.
```
drivers/flash/flash_stm32.c
drivers/flash/flash_stm32f4x.c
drivers/flash/flash_stm32l4x.c
```
Will extend STM32 support for nvs to enable test of missing files with `samples/subsys/nvs`.

vanwinkeljan, 2019-07-22:
Issued PR (https://github.com/zephyrproject-rtos/ci-dockerfiles/pull/44) to include SDL2 in CI image.
Once merged the SDL display driver could be covered via native_posix_64.

nashif, 2020-01-02:
closing this, the issue itself is not an enhancement, if we are to make an enhancement out of this, a new issue needs to be created, however I do not see what the enhancement is in this case.

closed 2020-01-02 by nashif