zephyrproject-rtos/zephyr #12886: Application development primer docs broken by west merge

mbolivar, 2019-01-30:
**Describe the bug**

The documentation recommends creating an application and build directory somewhere in the home directory. This is likely outside the west installation, so running commands like `ninja flash` or `west flash` in the build directory are unlikely to work.

**To Reproduce**

https://docs.zephyrproject.org/latest/application/application.html

**Expected behavior**

Docs should lead to working apps

**Impact**

Showstopper for v1.14; these docs must work

**Screenshots or console output**

E.g.:

"Create an application directory on your workstation computer, outside of the Zephyr base directory. Usually you’ll want to create it somewhere under your user’s home directory."

**Environment (please complete the following information):**
 - OS: N/A
 - Toolchain N/A
 - Commit SHA or Version used e.g. b73c9b32e8d84ad0a7820b5087db7cab681114f1

**Additional context**
N/A

carlescufi, 2019-01-30:
Well that is actually a major problem I had not foreseen. Do we need to require all external apps to be inside the zephyr installation? Can we make `ninja flash` work based on `ZEPHYR_BASE` and `$ZEPHYR_BASE..\.west` instead?

nashif, 2019-01-30:
i was surprised by this as well, discussed this briefly on #west yesterday.

mbolivar, 2019-01-30:
> Can we make `ninja flash` work based on `ZEPHYR_BASE` and `$ZEPHYR_BASE..\.west` instead?

I don't like the idea of using ZEPHYR_BASE because that could point to the wrong tree if users use zephyr-env.sh.

The west issue is https://github.com/zephyrproject-rtos/west/issues/175; please refer there for some discussion of the solutions.

SebastianBoe, 2019-01-31:
I don't know about "west flash".

But for "ninja flash", I would assume that since the flashing-related code is in-tree we could have some kind of in-tree frontend to invoke it. (Assuming we aren't able to solve "west flash").

mbolivar, 2019-02-01:
> But for "ninja flash", I would assume that since the flashing-related code is in-tree we could have some kind of in-tree frontend to invoke it. (Assuming we aren't able to solve "west flash").

Yeah, you read my mind. I'm going to try that. Monorepo "default" flashing and debugging can then continue to work.

mbolivar, 2019-02-01:
> Yeah, you read my mind. I'm going to try that. Monorepo "default" flashing and debugging can then continue to work.

This turns out to be impossible. The extension commands rely too much on modules inside west, such as `west.cmake`. So we really can't build this in-tree frontend.

closed 2019-02-05 by carlescufi