zephyrproject-rtos/zephyr #12905: Build improperly does a partial discard of 'const' defined variables

dleach02, 2019-01-31:
**Describe the bug**

The build system for Arm based targets performs improper discarding of ‘const’ defined variables that are only referenced within a single file. The map file shows the symbol in the Discard section but there is code still referencing that variable so the result is that the variable is located at address 0… all const variables that have this problem are located at address zero because the final linker stage didn’t fixup the address with a proper location. 

This problem can be seen in the ‘hello world’ sample by looking at what the linker does with _main_thread defined in init.c. The map file shows it as a discarded symbol but it is still in the elf file and you can inspect it with gdb and see that it is at address 0.

The final link stage links some of the libraries using the –whole-archive flag and some of the others using the –no-whole-archive flag so as an experiment I linked all libraries using just the whole-archive flag and it appeared to properly place the const variable. I'm not sure why our build uses these different flags so I'm not sure if doing this is an actual fix for the problem.

Also note that this doesn’t seem to be a problem when building for x86 targets.

**To Reproduce**

Build the 'hello world' sample application for an Arm based target (KW41, reel-board) and inspect how the const variable _main_thread (defined in init.c) is handled. It will appear in the Discard section in the linker map file. Use gdb to inspect the variable and to show the source address. It will show the address at 0.

**Expected behavior**
'const' variables need to be present in the rodata section and located appropriately .

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK)
 - Commit SHA or Version used

SebastianBoe, 2019-01-31:
I don't understand the problem generally or specifically for _main_thread.

Let's talk about specifically for _main_thread.

If _main_thread was placed in the .rodata section we would use more flash, so I am grateful that it does not. Why would we want to undo this optimization? _main_thread appears to have been inlined in a desirable way. Equivalent to this psuedo code:

```
int a = 3;
int * const a_ptr = &a;

main
{ 
  printk("%d", *a_ptr);
}

```

here it is not necessary to store a_ptr anywhere, because we already know before runtime what "*a_ptr" is going to become.

The same applies to _main_thread.

Again, I don't understand the problem generally, so it could be I am missing something.

dleach02, 2019-02-01:
@SebastianBoe, you need to separate _main_thread and its locality from the actual problem. 

The primary problem is that it is the linker's job to locate all symbols into the final image and then walk through all references to the symbols and fix up the underlying assembly instruction(s) to have the true address of the symbols. In the case of these const defined variables, the linker has decided that they should be discarded and as a result, it doesn't "fixup" any references to the symbols in the assembly language. Because the compiler generally emits an address of 0 it results in every const variable affected to be located at address zero plus what ever else was already located at zero. Also, the initial value that was assigned to the const symbol doesn't end up being there as well. This is an ugly problem.

dleach02, 2019-02-01:
The same problem exists in the 2.31.51.20181213 version of GNU tools.

andrewboie, 2019-02-01:
I really don't understand what the problem is.

If you define a const variable, and nothing ever uses it, then gc-sections is doing to discard it. Are you actually using these variables in some way that fools the linker?

dleach02, 2019-02-03:
@andrewboie, Yes the variable(s) is(are) still being used. The linker has improperly decided that these particular variables should be discarded so the linker doesn't go through its final fix up steps to allocate space and fix up the code locations that are referencing these variables with a proper address. Instead, the code uses what the compiler temporarily emitted which appears to be zero.

I highlighted _main_thread because it was an example that can easily be seen. Compile the "hello world" for an ARM architecture and you can see that the symbol is listed in the map file under the discarded sections. Then go in gdb and you can inspect the address of the symbol and see that it is at zero. If you look at the generated listing file you can see the various places it is referenced. You can build the same application for an x86 target and see that it properly allocates space for the symbol. I didn't go through all the different architectures but for now I think it is an ARM toolchain bug.

I narrowed this down to happening on 'const' defined variables that are only accessed from the file it is defined in. Now one could say that these variables should also be defined as 'static', which does correct the problem, but it is slightly annoying to have to search for these and fixup all instances of this pattern and further be aware of the bug in future code.

andrewboie, 2019-02-04:
> @andrewboie, Yes the variable(s) is(are) still being used.

I don't think you've proven this with your example.

Forgive me for being thick but if _main_thread is actually being used by the code, why isn't the application crashing?

_main_thread is defined in init.c, it's a pointer to _main_thread_s which is the actual thread struct, and nothing is actually using it, it appears that all use inside init.c is being optimized.

I bet you $0.02 that if you actually used _main_thread outside of init.c the linker wouldn't discard it.

dleach02, 2019-02-04:
I agree that if it is referenced outside of the file it is defined in then it is not discarded. This was what I pointed out in the initial problem report. 

But looking closer at the generated assembly code I can see that the "fixup" is PC relative and the value is the correct value... so likely this bug report is bogus. GDB is not helping here because when I inspect the variable it seems to be confused about its value.

dleach02, 2019-02-04:
After closer inspection of the generated assembly, this doesn't look like the bug I thought it was. GDB is still confused a bit but it looks like the generated assembly is correct.

closed 2019-02-04 by dleach02