zephyrproject-rtos/zephyr #12933: MCUboot: high current

chris-makaio, 2019-01-31:
When compiling our app "standalone" (flashed at offset 0) we see an average current of around 70uA at our PCA10056
But when enabling CONFIG_BOOTLOADER_MCUBOOT the average is around 500uA

Clock config is the same for both confs: RC with 20ppm

carlescufi, 2019-01-31:
This might be MCUboot initializing the UART. Can you try disabling logging and UART in MCUboot's Kconfig?

chris-makaio, 2019-02-01:
it's already disabled.
I guess it has something to do which the specs of Laird's BL652 when used w/o external crystal and using the internal 250ppm.
We'll add an external xtal, but it's still strange

carlescufi, 2019-03-05:
@chris-makaio since this is probably MCUboot or Laird-related, can you please close this?

closed 2019-03-05 by chris-makaio

AsgeirSH, 2019-09-13:
@chris-makaio Did you find out anything more about this? I'm seeing some of the same symptoms with higher current consumption on an nrf9160 dev kit, even with UART and logging disabled in MCUboot.

chris-makaio, 2019-09-13:
@AsgeirSH there are SO MANY possible issues related to power management - it's hard to tell what's involved. May I suggest you create a new issue, referencing this one? Are you sure it is related to MCUboot? Maybe it's an issue with that then