zephyrproject-rtos/zephyr #12935: Zephyr usurps "STRINGIFY" define

pfalcon, 2019-01-31:
If you'd need to define a macro which stringifies a symbol, how'd you call it? Surprise: everyone else calls it just the same. When different projects are used together, this leads to each redefine the other's, which leads to warnings, and with ever-popular -Werror, to build errors. And not even talking about cases where `#define STRINGIFY(x) itoa(x)`.

As anything else in Zephyr, STRINGIFY should be namespaced with prefix.

pfalcon, 2019-01-31:
~~~
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/plugins/ua_config_default.c:60:0: warning: "STRINGIFY" redefined
 #define STRINGIFY(arg) #arg
 
In file included from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/toolchain/gcc.h:31:0,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/toolchain.h:21,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/net/net_ip.h:28,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/net/socket.h:27,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/arch/posix/ua_architecture.h:29,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/BUILD.z/src_generated/ua_config.h:101,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/include/ua_types.h:19,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/include/ua_server.h:16,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/plugins/ua_config_default.h:12,
                 from /home/pfalcon/projects-3rdparty/Embedded/Zephyr/open62541/plugins/ua_config_default.c:13:
/home/pfalcon/projects-3rdparty/Embedded/Zephyr/zephyr/include/toolchain/common.h:100:0: note: this is the location of the previous definition
 #define STRINGIFY(s) _STRINGIFY(s)
 ~~~

Building https://github.com/open62541/open62541

pfalcon, 2019-01-31:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/12129

dleach02, 2020-06-16:
@pfalcon, can you provide a PR to do what you suggested?

nashif, 2020-07-25:
stale bug

closed 2020-07-25 by nashif