zephyrproject-rtos/zephyr #12959: Error with cmake build

vinny74man, 2019-02-01:
Hello everyone, after installing I tried with this example, "Blink_Led", but the result is this.

P.S. this is an online translation

sbsadmin@LAPTOP-T166O5K7:~/zephyrproject/zephyr/samples/basic/blink_led/build$ cmake -DBOARD=arduino_101 ..
Zephyr version: 1.13.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.7", minimum required is "3.4")
-- Selected BOARD arduino_101
/home/sbsadmin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: 1: /home/sbsadmin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc: Syntax error: "(" unexpected
CMake Error at /home/sbsadmin/zephyrproject/zephyr/cmake/host-tools.cmake:31 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /home/sbsadmin/zephyrproject/zephyr/cmake/app/boilerplate.cmake:378 (include)
  CMakeLists.txt:2 (include)

CMake Error at /home/sbsadmin/zephyrproject/zephyr/cmake/host-tools.cmake:32 (if):
  if given arguments:

    "VERSION_LESS" "1.4.6"

  Unknown arguments specified
Call Stack (most recent call first):
  /home/sbsadmin/zephyrproject/zephyr/cmake/app/boilerplate.cmake:378 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!

galak, 2019-02-01:
What version of the Zephyr SDK are you using?

Can you trying reporting results of:

`/home/sbsadmin/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc --version`

vinny74man, 2019-02-02:
Version: DTC 1.4.5

vinny74man, 2019-02-02:
I updated dtc to version 1.4.7, and now I get this.

sbsadmin @ LAPTOP-T166O5K7: ~ / zephyrproject / zephyr / samples / hello_world / build $ cmake -DBOARD = arduino_101 ..
CMake Error at CMakeLists.txt: 3 (include):
  includes could not find load file:

    /cmake/app/boilerplate.cmake

- The C compiler identification is GNU 7.3.0
- The CXX compiler identification is GNU 7.3.0
- Check for working C compiler: / usr / bin / cc
- Check for working C compiler: / usr / bin / cc - works
- Detecting C compiler ABI info
- Detecting C compiler ABI info - done
- Detecting C compile features
- Detecting C compile features - done
- Check for working CXX compiler: / usr / bin / c ++
- Check for working CXX compiler: / usr / bin / c ++ - works
- Detecting CXX compiler ABI info
- Detecting CXX compiler ABI info - done
- Detecting CXX compile features
- Detecting CXX compile features - done
CMake Error at CMakeLists.txt: 6 (target_sources):
  Can not specify sources for target "app" which is not built by this project.

- Configuring incomplete, errors occurred!
See also "/home/sbsadmin/zephyrproject/zephyr/samples/hello_world/build/CMakeFiles/CMakeOutput.log".

andrewboie, 2019-02-02:
>    /cmake/app/boilerplate.cmake

Did you source zephyr-env.sh?

vinny74man, 2019-02-03:
I run "source zephyr-env.sh" in the "/home/sbsadmin /zephyrproject /zephyr" dir

vinny74man, 2019-02-03:
Version of sdk 0.9.5

galak, 2019-02-03:
From the output you are reporting it seems like the `source zephyr-env.sh` didn't happen.  Can you try:

1. remove files in build dir
`rm -fr /home/sbsadmin/zephyrproject/zephyr/samples/hello_world/build/*`

2. source `zephyr-env.sh`
```
cd /home/sbsadmin/zephyrproject/zephyr`
source zephyr-env.sh
```

3. try building:
```
cd /home/sbsadmin/zephyrproject/zephyr/samples/hello_world/build
cmake -DBOARD=arduino_101 ..
```

nashif, 2019-02-19:
Please follow the getting started guide, you have been using the host compiler here, not the SDK.

closed 2019-02-19 by nashif