zephyrproject-rtos/zephyr #12969: settings: loading key-value pairs for given subtree

nvlsianpu, 2019-02-01:
Currently It is possible to load all the settings at once.
This make difficult design of big app when we rather want to initialize modules one by one. Especially when some modules should be initialized befor other  - which enforce split module initialization to initialization only for settings reception and further for the rest of initialization.

nvlsianpu, 2019-02-01:
cc @Qbicz Can you take the voice here?

nvlsianpu, 2019-02-01:
> But what if settings_load("") meaning load all is called after this, can the subtree be reloaded ?
Yes
> Or would we remove the settings_load("") completely and only allow settings_load("subtree") ?
No - keep API compatible backward.

Laczen, 2019-02-03:
I am still a little unsure of the expectations:
a. What if settings_load() is called before a settings_load_subtree() ?
b. Is the settings_load_subtree() limited to the settings.name or would you allow it to go down to the entry name ?

Qbicz, 2019-02-04:
@Laczen it should be possible to load whole tree or to load a chosen subtree. There are applications where it is beneficial to be able to load the settings during initialization of a module, but do not enforce that all other settings users are initialized at this point, letting modules not block each other in this process. But some applications will be happy will simplicity of loading everything at once.

I agree with @jhedberg it is up to application to load each subtree only once or know the consequences of reloading.

Laczen, 2019-02-04:
Hi @Qbicz, thanks for the elaboration. Maybe my concern is not exactly clear: suppose you have two modules where one is dependent (mod1) on the other (mod2), next to some other modules which do not care. If there is a routine settings_load() that loads all there is no way of ensuring this is not called before a settings_load_subtree(). This means that the mod1 set routine should catch the not yet being loaded of mod2 and decide to wait. Afterwards the mod1 would need to be loaded.
If we look at the settings as it is now this would mean we call settings_load() again. When settings_load_subtree() is available we can call settings_load_subtree("mod1").
So if we compare the advantage of adding the settings_load_subtree() it is very limited.
However there is a different way of getting a behavior that  is wanted by doing it differently: we define routines settings_load_exclude("subtree") that would set a filter on the set routines to exclude "subtree" from the set/commit routines. We also define a settings_load_include("subtree") to include it. Whenever we are ready with the filter we just call settings_load(). A call to settings_load() resets the filter to include all settings.

Qbicz, 2019-02-04:
@Laczen, in such case no one would load all: mod1 will do `settings_load_subtree(mod1)`, mod2 `settings_load_subtree(mod2)` and other module will load with `settings_load_subtree(othermodule)`. Currently BT module requires that application does `settings_load()` but IIRC this is in order to let application use settings also in other modules (because there is no subtree loading possible). After implementation of loading a subtree, I imagine BT stack can move to loading `settings_load_subtree("bt")`.

The only thing I'm worried about is if the BT stack moved to loading a subtree, application will be forbidden from calling `settings_load()`. In such case, excluding that you propose could be useful. @jhedberg what do you think?

nvlsianpu, 2019-02-04:
> b. Is the settings_load_subtree() limited to the settings.name or would you allow it to go down to the entry name ?

I think it should be possible to load subtree and certain key-value pair  as well.
I can also imagine that for some reason we can have a sub-tree and a key-value with the same name:
lest consider example of key stored:

`x/y`
`x/y/z`
`x/y/q`

The x/y is the key for some value and also the sub-tree for two children.
so when `settings_load_subtree("x/y")` will be called I expect that all keys will be loaded.

Laczen, 2019-02-04:
@Qbicz, your proposal would mean that settings_load() is removed and all modules should load calling their own subtree. This would break the backwards compatibility. It also means the settings file needs to be processed for each module.

@nvlsianpu, doing something like this is moving into the direction of a direct read of variables in the backend (and probably means it would be best handled at the backend for efficiency). The settings module seems like a very inefficient way to achieve that.

@Qbicz, @nvlsianpu, I think there needs to be a better understanding of what needs to be added. Creating a settings_load_subtree() only solves the dependency problem if settings_load() is removed, and results in multiple processing of the backend. Taking the load_subtree even lower than the module will result in even more processing of the backend.

My proposal would be:
a. limit the loading to a subtree and don't go lower.
b. extend the settings_handler with an "exclude" property that is standard set to "false" (so include it). Modules that need dependency set the exclude to true.
c. implement a settings_load_subtree() as: set all exclude to true, set exclude for the subtree to false, call settings_load().
d. the settings_handler "exclude" property is used in both the set and commit routines.

jhedberg, 2019-02-04:
> @Qbicz, your proposal would mean that settings_load() is removed and all modules should load calling their own subtree. This would break the backwards compatibility. 

Not if the settings subsystem tracks which modules have already been loaded (unless I misunderstood something with that proposal).

> It also means the settings file needs to be processed for each module.

Isn't that only true for the NFFS backend? Perhaps it would make sense to split the NFFS storage so that there's one file per module?

Laczen, 2019-02-04:
@jhedberg, I think the bt module is a good example of what is required:
a. You don't want the settings_load() to be called before everything is setup
b. Once everything is setup you want to call settings_load_subtree()
There are two requirements here, and only b is fulfilled with a settings_load_subtree().
So what if a user would like to change the bt settings (e.g. delete) before the bt module is setup ?

Laczen, 2019-02-04:
After going through the code I think the problem of dependency and requirement to wait until somethings are initialized can be done with just 1 line of code: in settings_handler_lookup(char *name) only return the handler if "enabled" .

So every settings_handler gets a bool "disabled" that is standard set to false.

If a module needs to wait for initialization or another module it declares the "disabled" property to true.

If initialization is finished or a required module is loaded the "disabled" property is set to false.

A settings_load() can be called at any moment and will not call the set/commit routines if a module is not ready.

Please comment if you see things that would not work with this method

nvlsianpu, 2019-10-23:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/16711 (`settings_load_subtree()` API)
even enhanced by https://github.com/zephyrproject-rtos/zephyr/pull/19541 + https://github.com/zephyrproject-rtos/zephyr/pull/19077 (`settings_load_subtree_direct()` API)

closed 2019-10-23 by nvlsianpu