zephyrproject-rtos/zephyr #12982: net: Zephyr drops IPv4 packet with extended MAC frame size

rmstoi, 2019-02-01:
If Zephyr receives a packet with IPv4 datagram occupying only portion of the MAC frame it drops the packet without responding to it. This is a regression from PR #11775.

As an example, attached also pcap from Linux, which responds correctly.

[ext-frame-icmp-zephyr.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2822242/ext-frame-icmp-zephyr.pcap.gz)
[ext-frame-icmp-linux.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2822243/ext-frame-icmp-linux.pcap.gz)

pfalcon, 2019-02-04:
> If Zephyr receives a packet with IPv4 datagram occupying only portion of the MAC frame it drops the packet without responding to it. 

I don't understand this report. Does it say that any IPv4 packet with the size different from 1500 gets dropped?

rmstoi, 2019-02-04:
@tbursztyka with PR #13024 UPD and ICMP work, but TCP fails, captures attached

[ext-frame-tcp-zephyr-pr12982.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2827457/ext-frame-tcp-zephyr-pr12982.pcap.gz)
[ext-frame-tcp-zephyr-05f41.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2827458/ext-frame-tcp-zephyr-05f41.pcap.gz)

tbursztyka, 2019-02-04:
Weird, how allocating incoming frame on ethernet driver affects tcp... will check.

rmstoi, 2019-02-04:
@pfalcon no, incoming packet has an ethernet trailer of 472 bytes, this causes failure in IP datagram size  calculations.

pfalcon, 2019-02-04:
> @pfalcon no, incoming packet has an ethernet trailer of 472 bytes, this causes failure in IP datagram size calculations.

In other words, there's a mismatch in IPv4 vs Ethernet packet sizes, and Ethernet packet has "empty" space after the IPv4 packet?

commit 968d423a (2019-02-04): net/pkt: Allocate relevant size on AF_UNSPEC

In case of Ethernet, if the requested size is larger than MTU and if
AF_UNSPEC is provided, the allocator will need to take into account
the ethernet header size which is not accounted in the MTU.

Other current L2 do not follow that rule as their MTU is based on IP
one (IPv6 most of the time). What they declare as MTU is the full frame
size they handle (minus the FCS for instance in 15.4). So with
AF_UNSPEC, such assumption on L2 header size is unrelevant.
(On 15.4 the header size is variable anyway and cannot be known until
the frame is parsed).
 [... cut]