zephyrproject-rtos/zephyr #12985: Add support for flash running over usb mass storage

galak, 2019-02-01:
Add a flash running that copies zephyr.bin over to the device across mass storage.  Ideally this would work cross platform, but also deal with issues like mounting, sync, etc.

jfischer-phytec-iot, 2019-02-01:
I hope it will not be the default runner for daplink based boards.

mbolivar, 2019-02-04:
> I hope it will not be the default runner for daplink based boards.

That's up to the board.cmake file. I certainly wouldn't want that for any boards I use regularly :).

With west, it's easy to `west flash -r cmsis-dap-mass-storage` (assuming the runner name is `cmsis-dap-mass-storage`; of course something shorter would be better, naming is hard)

jfischer-phytec-iot, 2019-02-04:
`west flash -r daplink-hic-msd` or just `west flash -r daplink-msd`?

mbolivar, 2019-02-06:
I like `daplink-msd` or maybe even just `usb-msd`.

jfischer-phytec-iot, 2019-02-06:
agree with usb-msd, I think some segger onboard adapters have it too.

galak, 2019-02-07:
I'm for 'usb-msd'

mbolivar, 2019-02-07:
SOLD at `usb-msd`

ioannisg, 2019-08-07:
@galak @mbolivar are we getting this into 2.0?

ioannisg, 2019-08-08:
Moving this to 2.1.0

dleach02, 2019-12-10:
We seem to be kicking this down the road. Does it make sense to to even put a milestone on this?

mbolivar, 2019-12-10:
@galak if you want this for 2.2 I can commit to doing it.

henrikbrixandersen, 2020-08-11:
I'd really like to see this runner make it to Zephyr. It would be very useful for boards, where the flash-algo is not supported by e.g. pyOCD.

mbolivar-nordic, 2020-10-05:
@henrikbrixandersen I'm going to drop the milestone as I don't have any plans for working on this at the moment.