zephyrproject-rtos/zephyr #12994: intermittent failures in tests/net/socket/select/ on qemu_x86

andrewboie, 2019-02-01:
**Describe the bug**
I'm intermittently seeing a problem with test_select on x86 where an assertion is failing. This happens often for me.

This seems to be a timing issue with QEMU if we get unlucky with a tick boundary. It depends on how much the workstation is loaded, unfortunately we don't have -icount working and QEMU tries to sync the target clock to the host clock.

Can we relax the zassert_true(tstamp <= 1, "") checks? In the failure case, tstamp has a value of 10. Are we trying to prove that the call isn't blocking?

**Observed Behavior**
```
SeaBIOS (version rel-1.11.2-0-gf9626ccb91-prebuilt.qemu-project.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.13.0-3961-gf2b6fca9b1 *****
Running test suite socket_select
===================================================================
starting test - test_fd_set
PASS - test_fd_set
===================================================================
starting test - test_select

    Assertion failed at /home/apboie/projects/zephyr3/tests/net/socket/select/src/main.c:97: test_select: (tstamp <= 1 is false)
got tstamp 10
FAIL - test_select
===================================================================
Test suite socket_select failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Expected behavior**
```
SeaBIOS (version rel-1.11.2-0-gf9626ccb91-prebuilt.qemu-project.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.13.0-3961-gf2b6fca9b1 *****
Running test suite socket_select
===================================================================
starting test - test_fd_set
PASS - test_fd_set
===================================================================
starting test - test_select
PASS - test_select
===================================================================
Test suite socket_select succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL
```

pfalcon, 2019-02-02:
> Are we trying to prove that the call isn't blocking?

Yes, that was the idea. For me, this issue doesn't seem to be reproducible, though while writing, I had to add `+ FUZZ` for a case below the one which fails for you. Of course, I can believe it may fail, and we should do something about it. I have following ideas:

1. Would you be interested to try `CONFIG_QEMU_TICKLESS_WORKAROUND=y`? It was in some other configs for timing-related tests, and I started with it too, but as I didn't see any visible effects, and wasn't sure what exactly it does, I decided to remove it.
2. We multiple existing timeouts in the test say by 3, to allow to still reliable distinguish "blocking" vs "non-blocking" behavior, and add `+ FUZZ` to "non-blocking" cases.
3. Can do whatever else quick patch you may suggest, e.g. just add `+ FUZZ` to the failing case.

pfalcon, 2019-02-04:
> I'd say add a system tick's worth of fuzz to each

Ok, then with adjusting timeouts for blocking calls, as then required to properly tell non-blocking from blocking, it's effectively p.3, and posted as https://github.com/zephyrproject-rtos/zephyr/pull/13028

PR #13028 (2019-02-04): tests: socket: poll, select: Fix non-blocking timing tests
The original test assumed that non-blocking select call would return
immediately, but actually it can be preempted by higher-priority
thread. So, we can account that it may possibly be delayed by a
scheduler quant of time (but otherwise we assume that testing
environment is light on other threads and their work, and one
quant is the most we can be delayed by).

Also, enable CONFIG_QEMU_TICKLESS_WORKAROUND. Most other
timing-related tests have this enabled, and it may affect
stability of QEMU testing too.

Fixes: #12994
commit 92962ddc (2019-02-04): tests: socket: poll, select: Fix non-blocking timing tests

As written originally, the tests assumed that if k_uptime_get_32()
returns times in milliseconds, that it also has millisecond
resolution. That however doesn't have to be the case, and indeed,
default timer interrupt period used by Zephyr is 10ms, and so system
time is incremented in such units. So, instead of "<= 1" tests to
account for timing increment, use "<= FUZZ".

For blocking tests, increase the timeout from previously used 10ms,
so we can reliably tests delays under the conditions described.

Also, enable CONFIG_QEMU_TICKLESS [... cut]
closed 2019-02-04 by nashif