zephyrproject-rtos/zephyr #12997: Make sanitycheck support multiple toolchains

nashif, 2019-02-02:
**Is your enhancement proposal related to a problem? Please describe.**
right now sanitycheck filters based on configured variant (from env variable). We should support multiple configured toolchains in the same run and filter based on a list rather than rely on the env variable.

**Describe the solution you'd like**
set variants on command line.

**Describe alternatives you've considered**
none

**Additional context**

In CI, we want to build with different toolchains, some boards can only be built with one toolchain and we need to cover those in CI in a consistent way.

alexandre-perrin, 2022-03-16:
I would like to suggest a more generic approach.
Instead of having a specific option for the toolchain, why not using an option for multiple environment sets? 

e.g. for testing with multiple toolchains:

```yml
tests:
  special.networking.test:
    env:
      ZEPHYR_TOOLCHAIN_VARIANT: 
      - gnuarmemb
      - zephyr
```

And on command line:

```
twister.py --multi_env ZEPHYR_TOOLCHAIN_VARIANT=[gnuarmemb,zephyr]
```

What do you think about it? As I am not very familiar with twister, it may not be relevant.

nashif, 2025-06-18:
that is already supported in different ways.