zephyrproject-rtos/zephyr #13003: context switch in x86 memory domains is terribly slow

andrewboie, 2019-02-03:
Context switching between threads in different memory domains on x86 is very slow with difficult to predict latency. Unlike other arches, which just write a fixed bank of registers, we have to go through and walk the page tables multiple times to rearrange the permissions on every page included by either the outgoing or incoming memory domain. This all happens with interrupts locked and can take varying amounts of time. This is very bad from a hard real-time perspective.

This is the code:

```
void _x86_swap_update_page_tables(struct k_thread *incoming,
				  struct k_thread *outgoing)
{
	/* Outgoing thread stack no longer accessible */
	_x86_mmu_set_flags((void *)outgoing->stack_info.start,
			   ROUND_UP(outgoing->stack_info.size, MMU_PAGE_SIZE),
			   MMU_ENTRY_SUPERVISOR, MMU_PTE_US_MASK);

	/* Userspace can now access the incoming thread's stack */
	_x86_mmu_set_flags((void *)incoming->stack_info.start,
			   ROUND_UP(incoming->stack_info.size, MMU_PAGE_SIZE),
			   MMU_ENTRY_USER, MMU_PTE_US_MASK);

	/* In case of privilege elevation, use the incoming thread's kernel
	 * stack, the top of the thread stack is the bottom of the kernel stack
	 */
	_main_tss.esp0 = incoming->stack_info.start;

	/* If either thread defines different memory domains, efficiently
	 * switch between them
	 */
	if (incoming->mem_domain_info.mem_domain !=
	   outgoing->mem_domain_info.mem_domain){

		 /* Ensure that the outgoing mem domain configuration
		  * is set back to default state.
		  */
		_arch_mem_domain_destroy(outgoing->mem_domain_info.mem_domain);
		_arch_mem_domain_configure(incoming);
	}
}
```

The x86 memory protection port is not intended for devices with constrained amounts of RAM. What we really need to do is when memory domains are initialized or modified, to allocate and configure a page table for that memory domain. Then upon context switch, instead of going through and doing expensive modification of a single page table with the two calls to _arch_mem_domain_destroy() and _arch_mem_domain_configure(), just update the CR3 register to point to the incoming thread's memory domain. This takes a fixed amount of time and is much faster.

The solution is not perfect as we still have to do some page table walking for adjusting the permission of thread stack pages for the incoming/outgoing threads, but this should still be an improvement. If we wanted to go all the way and make this completely a single step operation, we would need per-thread page tables which I feel is probably too expensive from a RAM perspective.

In more detail:

- The page table we generate at build time will be the kernel's page table, and will be used for any threads that are not part of any memory domain.

- We need to introduce a k_mem_slab pool of 4K pages, to draw from when we configure memory domains.

- When a memory domain is initialized, we will allocate some pages out of the slab pool and copy the kernel's page table into it, and save a pointer to that page table in some arch-specific field in the k_mem_pool struct (we will have to add this), since this page table is associated with the domain and not an individual thread.

- When a memory domain is configured, either at initialization or when partitions are added/removed, the domain's page table will be modified to suit the configuration. Removing partitions will just copy the old configuration for each affected page from the kernel's page table.

- On context switch:
1. We remove access to the outgoing stack like we currently do.
2. If the incoming thread is in a different domain, we load the incoming thread's domain page table into CR3.
3. We grant access to the incoming thread's stack memory like we currently do.

andrewboie, 2019-03-11:
Upon further contemplation:

- Think it's best to do page tables on a per-thread basis, not per-domain. This will have the lowest latency on context switch, for both stack and domain configuration it's just a quick update.
- We only need a set of PTEs per-thread for system RAM. For systems where memory starts at 0 (most x86), and have 2MB or less, this is one 4K page table. Context switch will just update its address in the PDE for it. If we have a nonzero origin, or more than 2MB, we might require more page tables for system ram, but the number of pages for each thread should be computable at build time.
- We already reserve a non-present page in every thread stack for the kernel mode stack overflow guard area. Dual purpose this. The thread-specific page table will go here, and the page will be marked read-only instead of non-present, so it still works as a guard page.
- Modification of thread page tables from supervisor mode will temporarily disable the WP bit in CR0 since the pages are read-only. If we don't do supervisor mode stack overflow detection we can skip this.
- Any modification of memory domains will require iterating over all the member threads and updating their page tables, but we should be able to do this in such a way to minimize amounts of time that interrupts are locked. We'll probably have to lock the scheduler during the entire operation.
- We should have infrastructure for performance profiling this.

andrewboie, 2019-04-10:
If we have per-thread page tables, the following actions need to take place when various memory domain APIs are called. This should also be applicable to #15223 

The following assumes that for any thread, HW state is applied when context switching in, or dropping to user mode.

- k_mem_domain_add_thread() / k_mem_domain_remove_thread() - Re-compute that thread's data to reflect the change in membership.
- k_mem_domain_destroy() - For all member threads, re-compute the thread's data to reflect no memory domain membership.
- k_mem_domain_init() - no thread actions
- k_mem_domain_add_partition() / k_mem_domain_remove_partition() - For all member threads, re-compute the thread's data to reflect the change in memory domain partitions

For re-computation on an MMU-based system, we need a baseline configuration, this will be copied from the kernel's page table, which has all the boot configuration applied.

We need two (and I think only two) arch APIs:

- z_arch_mem_domain_calculate(struct k_thread *thread) - For the given thread, examine its memory domain and update thread->arch data used to program the hardware. Invoked by the memory domain APIs when memory domains are changed, see above. May be a no-op if no thread->arch data is used.
- z_arch_mem_domain_configure(struct k_thread *thread) - Apply the thread's memory domain configuration to the MMU/MPU hardware. Called on context switch or drop to user mode.

Do we want an implicit call from, z_arch_mem_domain_calculate() to z_arch_mem_domain_configure() if the supplied thread == kernel->current?

wentongwu, 2019-04-11:
> If we have per-thread page tables, the following actions need to take place when various memory domain APIs are called. This should also be applicable to #15223
> 
> The following assumes that for any thread, HW state is applied when context switching in, or dropping to user mode.
> 
>     * k_mem_domain_add_thread() / k_mem_domain_remove_thread() - Re-compute that thread's data to reflect the change in membership.
> 
>     * k_mem_domain_destroy() - For all member threads, re-compute the thread's data to reflect no memory domain membership.
> 
>     * k_mem_domain_init() - no thread actions
> 
>     * k_mem_domain_add_partition() / k_mem_domain_remove_partition() - For all member threads, re-compute the thread's data to reflect the change in memory domain partitions
> 
> 
> For re-computation on an MMU-based system, we need a baseline configuration, this will be copied from the kernel's page table, which has all the boot configuration applied.
> 
> We need two (and I think only two) arch APIs:
> 
>     * z_arch_mem_domain_calculate(struct k_thread *thread) - For the given thread, examine its memory domain and update thread->arch data used to program the hardware. Invoked by the memory domain APIs when memory domains are changed, see above. May be a no-op if no thread->arch data is used. This can be time-consuming, but if at all possible, done without interrupts locked (locking the scheduler should be sufficient)
> 
>     * z_arch_mem_domain_configure(struct k_thread *thread) - Apply the thread's memory domain configuration to the MMU/MPU hardware. Called on context switch or drop to user mode. This API needs to be FAST.
>
 
I agree above design.

> Do we want an implicit call from, z_arch_mem_domain_calculate() to z_arch_mem_domain_configure() if the supplied thread == kernel->current?
>
It also doesn't take effect to the MPU(for arm) at once in current code if current thread change him-self's memory domain.

closed 2019-08-05 by carlescufi