zephyrproject-rtos/zephyr #13011: tests/posix/fs/ segfaults randomly in POSIX arch

aescolar, 2019-02-03:
**Describe the bug**
At random the test case will segfault just after passing in the POSIX architecture
(could only reproduce (easily) when compiling with COVERAGE enabled)

The issue is caused by the Zephyr POSIX sleep.c not being compiled with the posix_cheats.h defines.
When the POSIX arch attempted to call the host `sleep()`, instead the Zephyr POSIX API replacement sleep() was called, which messed up the threads state.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake ../tests/posix/fs/  -DBOARD=native_posix -GNinja  -DCONFIG_COVERAGE=y
3. ninja
4. (valgrind) zephyr/zephyr.exe
5. Admire a segfault in the posix arch most private code (should really not happen)

**Expected behavior**
No segfault

**Impact**
Intermitent segfault after the case pases => confused users (the likelyhood is not too high, so the 3 retries in CI were hiding it)

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr
 - Commit SHA: Todays master (69045011733b9e97bb) (probably started some time after 8dc69e09dad6366f2a443e37c7f3d5a7b3c4e2d9 )

**Additional context**
Bugfix incoming

closed 2019-02-03 by nashif

pfalcon, 2019-02-05:
@aescolar: I'd appreciate if I can be copied on all things POSIX subsys in Zephyr.

aescolar, 2019-02-05:
@pfalcon Certainly :)

PR #13012 (2019-02-03): arch: posix: Bugfix in POSIX cheats header
In 8dc69e09dad6366f2a443e37c7f3d5a7b3c4e2d9 (#10280)
the POSIX API main kconfig option was replaced from
PTHREAD_IPC to POSIX_API.
But the posix_cheats.h header was left using the old option.
This means that some applications/tests which were not
selecting the PTHREAD_IPC API, but using some other Zephyr
POSIX compatible APIs could have trouble when compiled for the
POSIX architecture.

Fixes #13011
commit 1305a9eb (2019-02-03): arch: posix: Bugfix in POSIX cheats header

In 8dc69e09dad6366f2a443e37c7f3d5a7b3c4e2d9 (#10280)
the POSIX API main kconfig option was replaced from
PTHREAD_IPC to POSIX_API.
But the posix_cheats.h header was left using the old option.
This means that some applications/tests which were not
selecting the PTHREAD_IPC API, but using some other Zephyr
POSIX compatible APIs could have trouble when compiled for the
POSIX architecture.

Fixes #13011