zephyrproject-rtos/zephyr #13025: CAN not working on Nucleo L432KC with external transciever

laperie, 2019-02-04:
**Describe the bug**
CAN sample (samples/drivers/CAN)  no longer works in any other mode than LOOPBACK.  In CAN_NORMAL mode, no signals are generated on CAN_RD and CAN_TD pins (PA_12 and PA_11 respectively).
The exactly similar environment works well with stm32f072b_disco but not with this one. 

**To Reproduce**
Build a CAN driver sample
Connect CAN transceiver 
Connect to another  supported device via CAN bus (e.g f072b_disco) and observe the behaviour

**Expected behavior**
L432KC board should receive messages from disco via can and send them upon button press

**Actual behavior**
1. Nothing is happening on the wire when button on Nucleo is pressed.
2. If user button is pressed on f072b_disco and CAN message is sent (as per logic analyzer) -- no interrupt is fired at the nucleo side, and the message is not received  

**Impact**
That's the only Zephyr board with CAN support that has >16K RAM. With that one broken we are very limited in advanced CAN-related works, such as SOCKET_CAN implementation

erwango, 2019-02-04:
@laperie, I don't have anything to test CAN.
Can you point out the commit that broke your configuration?

laperie, 2019-02-04:
@erwango   -- Cannot say for sure since I have never tested L432KC with anything else but LOOPBACK mode (that does not require 2nd device).  This mode continues to work, if that matters.

erwango, 2019-02-05:
@laperie, ok well this is important to know if we're looking for a regression or another issue, which could then be anything, including an issue related to L4 SoCs, this particular board, or even an issue in your test set up when using this particular board (In understood this board requires CAN transceiver and hence this is a bit different from using disco).
I'm editing the description to remove the "no longer works". It just doesn't work.

alexanderwachter, 2019-02-05:
@erwango I made a quick setup with F0 disco an L4 nucleo and it doesn't dork for me either.
Maybe I have time today to bring up a debug setup again to figure out whats wrong.

lowlander, 2019-02-05:
@laperie did you setup the pins correctly ? Not just in the DT but also in pinmux.c file of the board.  For the olimex p405 I did the following to make it work;

https://github.com/zephyrproject-rtos/zephyr/pull/12595/files#diff-3ea5c43750819a1f91789d273ceb61b2

alexanderwachter, 2019-02-05:
@lowlander This is already done in the standard board pinmux. I'm currently debugging. I think It's a problem witch the bs_1, bs_2 settings.

lowlander, 2019-02-05:
@laperie  where did you measure ? Really on the CAN bus L/H or on the GPIO pins of the CPU. Because if you switch rx/tx weird things happen.  (believe me I have tried :-)

alexanderwachter, 2019-02-05:
@lowlander we need to change the default device tree values to:
```
    bus-speed = <125000>;
    prop_seg_phase_seg1 = <4>;
    phase_seg2 = <5>
```
Otherwise we are not compatible to the nucleo and the prescaler  is not a natural number.
I did that but now i get
```
***** Booting Zephyr OS zephyr-v1.13.0-4079-g2b9458c378 *****
CAN: Device driver not found.
```

alexanderwachter, 2019-02-05:
Maybe I should make a dts overlay for different boards in the example. The problem is that prop_seg_phase_seg1 and phase_seg2 must be adjusted for different clock frequencies.

lowlander, 2019-02-05:
@alexanderwachter  I have a setup comparable to the linux can drivers, where I have min/max values for thinsg like SJW, phase1, phase2 in DT.  And tell the driver i wan 500kbit and 87.5% sample point an the driver figures things out itself.

alexanderwachter, 2019-02-05:
#13069 Fixes the problem. Tested against a stm32f072b_disco and logic analyzer with PulseView

alexanderwachter, 2019-02-05:
@laperie Maybe you have troubles with your transceiver too. The IO of your breakout board (http://skpang.co.uk/catalog/canbus-can-fd-breakout-board-5v-p-242.html) is 5V compatible but the nucleo has 3.3V IO. Also verify that you have 5V on the transceiver. Values slightly less than 5V won't work. Just had that problem. Don't connect to Vin but use the 5V pin.

closed 2019-02-05 by alexanderwachter