zephyrproject-rtos/zephyr #13050: net: Zephyr drops TCPv4 packet with extended MAC frame size

rmstoi, 2019-02-05:
If Zephyr receives a packet with IPv4 TCP datagram occupying only portion of the MAC frame it drops the packet without responding to it.

This regression seems to occur after commit 05f41 

Incorrectly working pcap (at master):
[ext-frame-tcp-zephyr-master.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2831461/ext-frame-tcp-zephyr-master.pcap.gz)

Correctly working pcap (at commit 05f41)
[ext-frame-tcp-zephyr-05f41.pcap.gz](https://github.com/zephyrproject-rtos/zephyr/files/2831451/ext-frame-tcp-zephyr-05f41.pcap.gz)

pfalcon, 2019-02-27:
What's the status of this?

tbursztyka, 2019-03-07:
That was fixed by 45df09c010d29684a0f2fe7e2b4751c9cea25b48  
closing

closed 2019-03-07 by tbursztyka