zephyrproject-rtos/zephyr #13051: Two timers are expiring at one time and crashing for platform nrf52_pca10040

kiranpachava, 2019-02-05:
**Describe the bug**
Two timers are expiring at one time and crashing for platform nrf52_pca10040
We have changed the expiry time to prime number or odd number.

**To Reproduce**
Steps to reproduce the behavior:
1. cd samples/hello_world
edit the file src/main.c and add the patch shared below.

--- a/samples/hello_world/src/main.c
+++ b/samples/hello_world/src/main.c
@@ -7,7 +7,27 @@
 #include <zephyr.h>
 #include <misc/printk.h>
 
-void main(void)
+void expiry_fn(struct k_timer *timer) {
+
+        printk("expired \n");
+}
+
+void stop_fn(struct k_timer *timer) {
+        printk("stoped \n");
+}
+
+K_TIMER_DEFINE(t1, expiry_fn, stop_fn);
+K_TIMER_DEFINE(t2, expiry_fn, stop_fn);
+
+ void main(void)
 {
-       printk("Hello World! %s\n", CONFIG_ARCH);
+       printk("Hello World! %s\n", CONFIG_ARCH);
+        printk("Hello World! %s\n", CONFIG_ARCH);
+
+        k_timer_start(&t1, 500, 200);
+        k_timer_start(&t2, 500, 400);
+
+        while (1) {
+                k_sleep(20000);
+        }

2. mkdir build; cd build
3. cmake -G Ninja -DBOARD=nrf52_pca10040
4. ninja 
5. ninja flash 
6. Launch the minicom (sudo minicom -D /dev/ttyACM0

**Expected behavior**

**Welcome to minicom 2.7.1

OPTIONS: I18n 
Compiled on May  3 2018, 15:20:11.
Port /dev/ttyACM0, 14:47:30

Press CTRL-A Z for help on special keys

***** Booting Zephyr OS zephyr-v1.13.0-25-g1630b5d3ac *****
Hello World! arm
Hello World! arm
expired 
***** HARD FAULT *****
  Fault escalation (see below)
***** MPU FAULT *****
  Instruction Access Violation
***** Hardware exception *****
Current thread ID = 0x20000cb8
Faulting instruction address = 0x20000ea0
Fatal fault in ISR! Spinning...**

**Impact**
when two timers are expired at same time app crashes.

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK 0.9.5, Branch code v1.13.0)

kiranpachava, 2019-02-06:
Hi,

Any update on the issue reported ? Please help to check it.

Regards,
Kiran

kiranpachava, 2019-02-08:
Hi,

Can someone help on this blocking issue ? 
Probably this will lead app crash to anyone who set the two timers in their programming.

Regards,
Kiran

andyross, 2019-02-26:
Taking a look.  I'm not able to reproduce this on my current tree which has a few unmerged fixes, so I suspect this is addressed.  Will check against master.

kiranpachava, 2019-02-26:
Please do check it on master branch code (v1.13.0 release tag).

andyross, 2019-02-26:
Oh, was this against the 1.13 release code?  Can you please check against master?   If it's fixed it's fixed, and we can close this.

kiranpachava, 2019-02-26:
yep, reported on v1.13 release code. Sure we will check this after merging v1.14.0-rc1 to our code base. 
Thanks Andy, I will get back to you on this if we find any issue on latest code base.

andyross, 2019-02-26:
To be clearer: basically the entire timer subsystem was rewritten for 1.14.  And nRF in particular was subject to a bunch of timer bugs with the old code.  The new code has 100% new and improved bugs.

kiranpachava, 2019-02-27:
Thanks Andy, I checked it and the issue is not reproduced in v1.14.0-rc1 code base

andyross, 2019-02-27:
Fantastic, thanks.

closed 2019-02-27 by andyross