zephyrproject-rtos/zephyr #13075: Review memory protection Kconfig policies for consistency and sanity

andrewboie, 2019-02-05:
We have a number of Kconfigs related to memory protection which I think could use some attention and or clarification.

- Do we want to support memory domains for arches that do not have userspace enabled? This seems inconsistent; for example on x86 memory domains don't work with userspace disabled, but it seems like on ARM that they do. If we do support with userspace disabled, what are the use-cases. Right now the kernel is not designed to limit its own access to system RAM. My inclination is to, at least until we have a design for restricting supervisor mode access to memory, to make it part of userspace, but we can discuss it.

- I think CONFIG_APP_SHARED_MEM config should be removed, and the app shared memory feature rolled into userspace itself, or at least memory domains. With a little tweaking we can make the footprint/performance penalty for this feature zero if no app shared memory partitions are defined.

- We need to make sure that on all arches, CONFIG_USERSPACE works with CONFIG_HW_STACK_PROTECTION disabled. It saves RAM for people who don't care about catching stack overflows in kernel mode, which is a valid use-case as such overflows are not guaranteed to be recoverable and it will free an MPU region and space for the guard area.

- I think there are some inconsistencies among arches on what the semantics of STACK_ALIGN are.

ruuddw, 2019-02-06:
I see a (strong) case for MPU protection in non-userspace builds (e.g. code/execution protection), but your concern is more specifically on memory domains for kernel, right?
My interpretation/assumption for memory domains matches the "ARM" version, I think as mentioned before: principle of least privilege, no access if not needed.

Agree on the CONFIG_APP_SHARED_MEM - it will be hard to build userspace applications without it.

CONFIG_HW_STACK_PROTECTION indeed is othogonal to userspace. What confuses this even more is the HW stack bounds checking, and MPU based stack guards. Same goals, different implementation and protection details.

What are the STACK_ALIGN concerns?

ioannisg, 2019-06-30:
> Do we want to support memory domains for arches that do not have userspace enabled? This seems inconsistent; for example on x86 memory domains don't work with userspace disabled, but it seems like [...]

For ARM we do not have memory domains if user mode support is not enabled. But we already use memory partitions (k_mem_partition) to enforce non-usermode-related properties, i.e. no-cache, ram-func, read-only, etc.

 >I think CONFIG_APP_SHARED_MEM config should be removed, and the app shared memory feature rolled into userspace itself, or at least memory domains. With a little tweaking we can make the footprint/performance penalty for this feature zero if no app shared memory partitions are defined.

This has already been removed, right?

> We need to make sure that on all arches, CONFIG_USERSPACE works with CONFIG_HW_STACK_PROTECTION disabled. It saves RAM for people who don't care about catching stack overflows in kernel mode, which [...]

For ARM this is already the case; HW_STACK_PROTECTION and USERSPACE are non dependent on each other.

>I think there are some inconsistencies among arches on what the semantics of STACK_ALIGN are. ARC definitely seems to be doing something different.

Perhaps. But for ARM we already re-worked these macros and defined, now, a complete set of alignment macros for all different cases:
- thread stack align
- mem partition align
- privilege stack align

ioannisg, 2019-06-30:
IMHO, ARM architecture already addresses most of these points. Could we either close this @andrewboie @ruuddw or narrow it down to more fine-grained topics?

ruuddw, 2019-07-01:
I think these indeed have been addressed. Though still not sure what the STACK_ALIGN interpretation difference for ARC was. Can be closed imho.

closed 2019-07-01 by andrewboie