zephyrproject-rtos/zephyr #13113: Samples fail to build for SimpleLink when CONFIG_XIP=n

vanti, 2019-02-06:
**Describe the bug**
When trying to build the samples for cc3220xl_launchxl with the code placed in SRAM using CONFIG_XIP=n, I see the build fail:

```
[ 44%] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
In file included from /home/vwan/zephyr-work/include/toolchain.h:21:0,
                 from /home/vwan/zephyr-work/samples/basic/disco/build/zephyr/misc/generated/configs.c:3:
/home/vwan/zephyr-work/samples/basic/disco/build/zephyr/misc/generated/configs.c: In function ‘_ConfigAbsSyms’:
/home/vwan/zephyr-work/include/toolchain/gcc.h:341:51: error: expected expression before ‘)’ token
   "\n\t.type\t" #name ",%%object" :  : "n"(~(value)))
                                                   ^
/home/vwan/zephyr-work/samples/basic/disco/build/zephyr/misc/generated/configs.c:44:1: note: in expansion of macro ‘GEN_ABSOLUTE_SYM’
 GEN_ABSOLUTE_SYM(CONFIG_FLASH_SIZE, );
 ^~~~~~~~~~~~~~~~
/home/vwan/zephyr-work/samples/basic/disco/build/zephyr/misc/generated/configs.c:45:45: error: invalid suffix "x" on integer constant
 GEN_ABSOLUTE_SYM(CONFIG_FLASH_BASE_ADDRESS, 0x);
                                             ^
/home/vwan/zephyr-work/include/toolchain/gcc.h:341:46: note: in definition of macro ‘GEN_ABSOLUTE_SYM’
   "\n\t.type\t" #name ",%%object" :  : "n"(~(value)))
                                              ^~~~~
zephyr/CMakeFiles/zephyr.dir/build.make:218: recipe for target 'zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj' failed
make[2]: *** [zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj] Error 1
CMakeFiles/Makefile2:536: recipe for target 'zephyr/CMakeFiles/zephyr.dir/all' failed
make[1]: *** [zephyr/CMakeFiles/zephyr.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior using the blinky or disco sample:
1. Add CONFIG_XIP=n to prj.conf
2. mkdir build; cd build
3. cmake .. -DBOARD=cc3220sf_launchxl
4. make
5. See error

**Expected behavior**
I expect the build to be successful.

**Impact**
Prevents anyone from placing code into SRAM using CONFIG_XIP=n. Must leave the code in flash.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.9.5
 - Commit SHA where this was reproduced: a10f07305fce821183762d624150cc641866dd70

vanti, 2019-02-06:
@galak  Commit `ce7ed18989e0c03ab06102da3c321247f775337e dts: Move CONFIG_SRAM_* & CONFIG_FLASH_* back to Kconfig` seems to have added some new Kconfig options. Are we expected to do something extra to support them in SimpleLink?

GAnthony, 2019-02-07:
@galak, added the 1.14 label, as this does affect our normal flow of development/debugging.
Also, what's the best way to add a test case which sets CONFIG_XIP=n, so it can trigger a build failure for cc3220sf_launchxl board in the CI build tests?

galak, 2019-02-07:
Take a look at commit d1e55eb842bbed53a0ebee58d1d8711a480f1c2d, let's see if doing something similar fixes the issue.

GAnthony, 2019-02-07:
@galak, Thanks for the guidance.  
Submitted
https://github.com/zephyrproject-rtos/zephyr/pull/13138

PR #13138 (2019-02-07): boards: cc3220sf_launchxl: Set CONFIG_FLASH defaults for non-XIP
In case of !XIP, set defaults for CONFIG_FLASH_SIZE and
CONFIG_FLASH_BASE_ADDRESS in Kconfig.

Fixes: #13113
commit 8b457d20 (2019-02-07): boards: cc3220sf_launchxl: Set CONFIG_FLASH defaults for non-XIP

In case of !XIP, set defaults for CONFIG_FLASH_SIZE and
CONFIG_FLASH_BASE_ADDRESS in Kconfig.

Fixes: #13113
closed 2019-02-07 by galak