zephyrproject-rtos/zephyr #13114: net subsystem needs documentation on how net_pkt and net_buf can be used

aurel32, 2019-02-06:
The net subsystem uses `net_pkt` and `net_buf` intensively. Both are well documented, however documentation on how they have to be used, ie what assumptions can be done and what precautions are needed before changing a packet, is clearly lacking. For instance depending if the packet is in the RX path or the TX path, at L2 or L3 level, the assumptions and requirements are not the same.

Such a documentation would be useful for these PR and probably many others:
https://github.com/zephyrproject-rtos/zephyr/pull/12564#issuecomment-460965844
https://github.com/zephyrproject-rtos/zephyr/pull/12563

It would also be a way to make sure everybody agree about how the IP stack is designed and maybe a way to find possible issues in the current code or design.

jukkar, 2019-02-19:
This is related to #13378 which talks about enhancing API documentation.
The  zephyr/doc/reference/networking/net_pkt.rst file should have more information about usage of net_pkt in it.
I am assigining this to @tbursztyka as he has worked on this recently and knows the API usage best atm.

tbursztyka, 2019-03-18:
working on it.
@nashif can it still make it for 1.14?

commit a10df124 (2019-03-25): doc/networking: Add documentation on struct net_pkt and its API

Memory managment and operations.

Fixes #13114
closed 2019-03-25 by nashif