zephyrproject-rtos/zephyr #13122: dts: build for KW40z, KW41z fails to generate isr_tables

jremmert-phytec-iot, 2019-02-07:
**Describe the bug**
Building an example for a frdm_kw41z results in a failure in generating isr_tables when gpio portb is enabled:
```
[ 97%] Generating isr_tables.c
gen_isr_tables.py: multiple registrations at table_index 31 for irq 31 (0x1f)
Traceback (most recent call last):
  File "/[..]/zephyr/arch/common/gen_isr_tables.py", line 308, in <module>
    main()
  File "/[..]/zephyr/arch/common/gen_isr_tables.py", line 300, in main
    error("multiple registrations at table_index %d for irq %d (0x%x)" % (table_index, irq, irq))
  File "/[..]/zephyr/arch/common/gen_isr_tables.py", line 35, in error
    raise Exception()
Exception
make[2]: *** [zephyr/CMakeFiles/kernel_elf.dir/build.make:63: zephyr/isr_tables.c] Error 1
make[1]: *** [CMakeFiles/Makefile2:314: zephyr/CMakeFiles/kernel_elf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:
1. enable GPIO_MCUX_PORTB in boards/arm/frdm_kw41z/Kconfig.defconfig
2. build e.g. hello_world example
3. cmake -DBOARD=frdm_kw41z
4. make
5. See error

**Impact**
Examples for frdm_kw41z or boards based on kw40z and kw41z will fail in building when all gpios are activated

**Screenshots or console output**
in debug mode, I'm getting this:
```
[ 97%] Generating isr_tables.c
gen_isr_tables.py: (32, 0)
gen_isr_tables.py: Configured interrupt routing
gen_isr_tables.py: handler    irq flags param
gen_isr_tables.py: --------------------------
gen_isr_tables.py: 0x1957     31  0     0x20000ed0
gen_isr_tables.py: 0x1957     31  0     0x20000edc
gen_isr_tables.py: 0x1957     30  0     0x20000ee8
gen_isr_tables.py: offset is 0
gen_isr_tables.py: num_vectors is 32
gen_isr_tables.py: multiple registrations at table_index 31 for irq 31 (0x1f)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 0f3bd0f43bf5bbc559303dc91d4fc3aacb58281e

**Additional context**
This check was recently introduced by #11273 by @andrewboie, issue is similar to k64 in #11275.
possible fix: Assign a different interrupt number in nxp_kw40z.dtsi and nxp_kw41z.dtsi.
How were the numbers 30/31 selected? Which number to select instead of 31?

galak, 2019-02-07:
I think the change from @andrewboie is just showing an issue in that we need to implement a shared GPIO handler for this case in the mcux gpio driver.

andrewboie, 2019-02-07:
```
gen_isr_tables.py: 0x1957     31  0     0x20000ed0
gen_isr_tables.py: 0x1957     31  0     0x20000edc
```

Yeah, before I added the check it was undefined which would be installed; same handler function in both cases but the parameter would be one or the other for all interrupts which is not what you want.

agansari, 2019-02-18:
@jremmert-phytec-iot KW41z platform uses IRQ 31 for both Port B & C's GPIOs. Port B is also used for ADC and other peripherals, this is why it's normally disabled.
@galak @andrewboie a proper implementation would be to check on IRQ 31 both PORTB_ISFR and PORTC_ISFR registers to check the source of the interrupt in this particular case. In general, I don't see how to create a generic configuration for peripherals that share the same interrupt.

A fast and middle-ground way to fix this is to remove interrupts on GPIOB.
GPIO on port B is not used by default, no pinmux initialization; let alone pin interrupts.

PR #15127 (2019-04-03): dts: nxp: Disable kw40/41 gpiob interrupts
Ports B and C share a common interrupt vector on kw40 and kw41z socs,
but we don't currently have a way to express this in device tree. A
check was added in commit 77cb942a976a811c0da4aaaa57d75e2915c104bc that
correctly causes build errors on kw40/41 boards when both ports are
enabled.

Disable the port b interrupt for now until we have a better way to
handle this.

Fixes #13122. Alternative to #14653, which fixes the build error but doesn't actually work at runtime due to `gpio_mcux_configure` returning an error code.
closed 2019-04-03 by nashif