zephyrproject-rtos/zephyr #13146: net: RX packets are sometimes used in the TX path

aurel32, 2019-02-07:
If we want to have a sane IP stack, we should not allow RX packets (and thus RX buffers) to be used in the TX path. Or if we allow that, we should just use a common pool for both TX and RX. This just makes additional constraints on the drivers and increases the minimum RX buffer requirements.

This happens where `net_pkt_copy()` or `net_pkt_copy_all()` are used, and most notably the icmpv4 and icmpv6 code. A possible fix is to pass the slab to use as an argument of those functions.

pfalcon, 2019-02-07:
> Or if we allow that, we should just use a common pool for both TX and RX.

So, we exactly want to allow to have a dedicated, even adhoc RX pools. E.g., pools in memory-mapped NIC controller memory, which can be used for nothing else but RX packets (and which, being overflowed, leads to problems, so we should not hold onto packets from that pool for long, like send the TX path, whatnot).

Thanks for submitting this ticket!

cc @jukkar, @tbursztyka

tbursztyka, 2019-02-08:
Actually, that issue does not exist anymore. I changed the core while applying the new net_pkt API so it would not reuse the RX pkt to TX. (originally icmpv4 and v6 in their respective echo request handler)

There are indeed legacy function still (like net_pkt_copy/net_pkt_copy_all) but these are not used in the core anymore. Actually I could remove these 2 in my PR #12989 

Besides that, I do have plans - old ones actually - to use only 1 unique pool for all. The split was originally made for wrong reasons.

aurel32, 2019-02-08:
Indeed it is now fixed, sorry. I encountered the issue 2 weeks ago when working on the GMAC driver, and since them kept on my TODO list to submit a bug. Sorry for not checking properly the new code, I didn't notice the `_new` version of `net_pkt_copy`.

It means we can now revert https://github.com/zephyrproject-rtos/zephyr/commit/afbee4c96abedfdca842e5a7be0330f2ca8575

closed 2019-02-08 by aurel32

tbursztyka, 2019-02-08:
@aurel32 no worry, the changes I brought are quite broad, it changes a lot of things packet flow in the net stack. this issue helped me to notice net_pkt_clone/net_pkt_copy legacy functions could be removed actually.