zephyrproject-rtos/zephyr #13148: Build on nucleo_stm32f429zi

vinny74man, 2019-02-07:
Hello everyone, I'm trying the example "blink_led" for a stm32f429zi I can not perform make returns various errors can you help me to set the various parameters for this board?
Thank you

P.S. this is an online translation, sorry

These are the errors:

[  8%] Building C object CMakeFiles/app.dir/src/main.c.obj
/home/sbsadmin/zephyrproject/zephyr/samples/basic/blink_led/src/main.c:45:2: error: #error "Choose supported PWM driver"
 #error "Choose supported PWM driver"
  ^~~~~
/home/sbsadmin/zephyrproject/zephyr/samples/basic/blink_led/src/main.c: In function ‘main’:
/home/sbsadmin/zephyrproject/zephyr/samples/basic/blink_led/src/main.c:62:31: error: ‘PWM_DRIVER’ undeclared (first use in this function)
  pwm_dev = device_get_binding(PWM_DRIVER);
                               ^~~~~~~~~~
/home/sbsadmin/zephyrproject/zephyr/samples/basic/blink_led/src/main.c:62:31: note: each undeclared identifier is reported only once for each function it appears in
/home/sbsadmin/zephyrproject/zephyr/samples/basic/blink_led/src/main.c:69:33: error: ‘PWM_CHANNEL’ undeclared (first use in this function)
   if (pwm_pin_set_usec(pwm_dev, PWM_CHANNEL,
                                 ^~~~~~~~~~~
CMakeFiles/app.dir/build.make:62: recipe for target 'CMakeFiles/app.dir/src/main.c.obj' failed
make[2]: *** [CMakeFiles/app.dir/src/main.c.obj] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/app.dir/all' failed
make[1]: *** [CMakeFiles/app.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

galak, 2019-02-08:
Without letting us know which board you are building for, its hard to say.  If its a board in the Zephyr tree we can look at this in more detail.  If its not, than its mostly due to not having PWM configured for that board.

vinny74man, 2019-02-08:
The board is the, NUCLEO STM32F429Zi, and is supported

erwango, 2019-02-08:
As you can see, at beginning of samples/basic/blink_led/src/main.c, there is a section of code in which we do the needed tuning to adapt the test case to the board. You need to tune that to make test working on this test.

```
#if defined(CONFIG_SOC_STM32F401XE) || defined(CONFIG_SOC_STM32F412ZG) || \
	defined(CONFIG_SOC_STM32F413XX) || defined(CONFIG_SOC_STM32L476XG) || \
	defined(CONFIG_SOC_STM32F407XG) || defined(CONFIG_SOC_STM32F302X8)
#define PWM_DRIVER DT_PWM_STM32_2_DEV_NAME
#define PWM_CHANNEL 1
#elif CONFIG_SOC_STM32F103XB
#define PWM_DRIVER DT_PWM_STM32_1_DEV_NAME
#define PWM_CHANNEL 1
...
```

vinny74man, 2019-02-08:
I saw the definitions, so I asked for help with the configuration of my board, I can not figure out how to configure the main settings.

Thank you

P.S. this is an online translation, sorry

vinny74man, 2019-02-10:
Hello everyone, I managed to run build, with example basic / blinky, on stm32f429zi, now my question is how do I configure a pin different from the default?
In the sense if I want the PB5 as an output pin in this example:

#define LED_PORT LED0_GPIO_CONTROLLER
#define LED LED0_GPIO_PIN
This enables the PB0 which is the led0 on-board, how controller should I write?
I tried with GPIOB, STM_32_GPIOB, but it does not work.
Where do I get this information, in which part of the guide?

Thank you

P.S. this is an online translation, sorry

erwango, 2019-02-11:
LED0_GPIO_CONTROLLER/PIN is generated from board .dts file, which is the device tree board description. It is generated from led0 alias defined in this file.
You can modify this alias or add a new one.
For more information on device tree, cf https://www.devicetree.org/

vinny74man, 2019-02-11:
But I do not want to use alias I want to insert port and pin, can you do?
I would like to do this in src / main.c

#define Led_R_Driver GPIOB
#define Led_R_Pin 0

What should I write in place of GPIOB, GPIO ports as they are identified, or in which files are they defined?

Thank you

P.S. this is an online translation, sorry

erwango, 2019-02-13:
You can try:
```
	dev = device_get_binding("GPIOB");
```

vinny74man, 2019-02-13:
Yes thanks, I understood that the function "device_get_binding" required a string, and I tried, now I tried the same thing for the example "blink_led", but it does not work, what are the correct parameters for PWM controllers?

Thank you.

P.S. this is an online translation, sorry

erwango, 2019-02-14:
For PWM_2 it will be ```"PWM_2"```.
You should actually use the label defined in dts nodes (cf `dts/arm/st`).

vinny74man, 2019-02-14:
OK thanks.

vinny74man, 2019-02-14:
It does not work, I'm trying the PB7 pin that corresponds to the TIM4 cahnnel 2, so I use it
in /src/main.c

#elif CONFIG_SOC_STM32F429XI
#define PWM_DRIVER "PWM_4",
#define PWM_LED 7

But it does not work, starting the program on the serial I get this
  ***** Booting Zephyr OS zephyr-v1.13.0-4563-g0e4ff809d7 *****
PWM demo app-blink LED
Can not find PWM_4!
Thank you.

P.S. this is an online translation, sorry

erwango, 2019-02-15:
You should also enable PWM_4 instance with Kconfig:
In `boards/arm/nucleo_f429zi/nucleo_f429zi_defconfig`:
```
CONFIG_PWM_STM32_4=y
```
Also, you should configure the right pin in `boards/arm/nucleo_f429zi/pinmux.c`.

vinny74man, 2019-02-15:
Ok, so for every pin that is not by default, I have to change this file:
"Boards / arm / nucleo_f429zi / nucleo_f429zi_defconfig"
that this:
boards / arm / nucleo_f429zi / pinmux.c

Thank you

P.S. this is an online translation, sorry

erwango, 2019-02-15:
@vinny74man , please let us know when you're able to build so we can close this point.
Or you can do it whenever you want.

vinny74man, 2019-02-15:
Ok

vinny74man, 2019-02-15:
Does not work.
I modified the file /src/main.c by adding these lines:

#elif CONFIG_SOC_STM32F429XI
#define PWM_DRIVER "PWM_4"
#define PWM_CHANNEL 2

the file /arm/nucleo_f429zi/nucleo_f429zi_defconfig in this way:

# enable PWM_4
CONFIG_PWM_STM32_4 = y

the file boards/arm/nucleo_f429zi/pinmux.c in this way:

#ifdef CONFIG_PWM_STM32_4
         {STM32_PIN_PB7, STM32F4_PINMUX_FUNC_PB7_PWM4_CH2},
          #endif / * CONFIG_PWM_STM32_4 * /
 
but the serial always prints this:

***** Booting Zephyr OS zephyr-v1.13.0-4563-g0e4ff809d7 *****
PWM demo app-blink LED
Can not find PWM_4!

Thank you

P.S. this is an online translation, sorry

erwango, 2019-02-18:
Works fine on my side.
On following branch you'll see the changes required to get the sample working:
https://github.com/erwango/zephyr/tree/blink_led_nucleo_f429zi

erwango, 2019-04-23:
@vinny74man, did you manage to get it working?
If yes, can we close this ticket?

erwango, 2019-07-23:
Closing ticket since no response and tested ok.

closed 2019-07-23 by erwango