zephyrproject-rtos/zephyr #13151: Update documentation on linking Zephyr within a flash partition

mnkp, 2019-02-07:
PR #12108 "Shift DTS before Kconfig" modified how flash partitions are linked. Before #12108 was merged the process was as follows:
> If the chosen node has no zephyr,code-partition property, the application image link uses the entire flash device. If a zephyr,code-partition property is defined, the application link will be restricted to that partition.

this is no longer true.

Following documentation needs to be updated:
- "[Linking Zephyr Within a Partition](https://docs.zephyrproject.org/latest/devices/dts/flash_partitions.html#linking-zephyr-within-a-partition)"
- the help text of BOOTLOADER_MCUBOOT Kconfig option in ./Kconfig.zephyr
- majority of board dts files that define flash partitions `find boards/ -name "*.dts" | xargs grep -l partitions` document linking process within flash node.

ioannisg, 2019-02-08:
Yeah, now it is more like all nodes have the code-partition, somehow defined in DTS of the boards, but it is only used if there is no BOOTLOADER, correct?

mnkp, 2019-02-08:
It seems like the flash partition layout and setting of zephyr,code-partition property is taken into account only if BOOTLOADER_MCUBOOT Kconfig option is set.

psidhu, 2019-04-25:
This affects adafruit_trinket_m0. It's required that the adafruit_trinket_m0 boots from 0x2000 due to their UF2 bootloader.

zephyr,code-partition isn't taken into account because BOOTALODER_MCUBOOT isn't set apparently. So currently, no one can boot zephyr on the adafruit_trinket_m0 unless they physically change where flash begins e.g. something like:

```
	soc {
		nvmctrl: nvmctrl@41004000  {
			flash0: flash@0 {
				reg = <0x2000 0x3E000>;
			};
		};
	};
```

I'm sure there are other boards affected by this.

sslupsky, 2019-11-25:
See my comments PR #20740  and the documentation about flash partitions here https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/guides/dts/flash_partitions.inc

The Adafruit board dts is correct.  What is missing is the defconfig should include

```CONFIG_USE_CODE_PARTITION=y```

PR #22919 (2020-02-19): Update documentation related to linking the code partition
The method used to link code partition, as defined by zephyr,code-partition has been modified in Zephyr 1.14. Update the "Linking Zephyr Within a Partition" section to reflect the change. Also remove any remaining outdated documentation embedded in board dts files.

Update description of BOOTLOADER_MCUBOOT Kconfig option.

This fixes #13151 in master.
closed 2020-02-19 by galak