zephyrproject-rtos/zephyr #13166: tests/kernel/threads/dynamic_thread test cases are failing on frdm_k64f board

arun1joshi, 2019-02-08:
The tests in tests/kernel/threads/dynamic_thread fail with Assertion failure on frdm_k64f board
Note: the failure looks similar to the closed issues #10475, #10476

The test started failing from this commit: [a10f073]

**Arch**: arm
**Platforms**: frdm_k64f

**Error on console log**:
frdm_k64f :arm_
***** Booting Zephyr OS zephyr-v1.13.0-4309-g0a7b45d55f (delayed boot 1000ms) *****
Running test suite thread_dynamic
===================================================================
starting test - test_kernel_create_dyn_user_thread
***** BUS FAULT *****
  Imprecise data bus error
  NXP MPU error, port 3
    Mode: Supervisor, Data Address: 0x20000d24
    Type: Write, Master: 0, Regions: 0x8100
***** Hardware exception *****
Current thread ID = 0x200001e8
Faulting instruction address = 0x313a
Fatal fault in thread 0x200001e8! Aborting.

    Assertion failed at zephyr.git/tests/kernel/threads/dynamic_thread/src/main.c:51: create_dynamic_thread: (k_sem_take(&end_sem, K_SECONDS(1)) == 0 is false)
k_sem_take(end_sem) failed
FAIL - test_kernel_create_dyn_user_thread
===================================================================
starting test - test_user_create_dyn_user_thread
***** BUS FAULT *****
  Imprecise data bus error
  NXP MPU error, port 3
    Mode: Supervisor, Data Address: 0x20000d24
    Type: Write, Master: 0, Regions: 0x8100
***** Hardware exception *****
Current thread ID = 0x200001e8
Faulting instruction address = 0x313a
Fatal fault in thread 0x200001e8! Aborting.

    Assertion failed at zephyr.git/tests/kernel/threads/dynamic_thread/src/main.c:51: create_dynamic_thread: (k_sem_take(&end_sem, K_SECONDS(1)) == 0 is false)
k_sem_take(end_sem) failed
FAIL - test_user_create_dyn_user_thread
===================================================================
starting test - test_dyn_thread_perms
***** BUS FAULT *****
  Imprecise data bus error
  NXP MPU error, port 3
    Mode: Supervisor, Data Address: 0x20000d24
    Type: Write, Master: 0, Regions: 0x8100
***** Hardware exception *****
Current thread ID = 0x200001e8
Faulting instruction address = 0x313a
Fatal fault in thread 0x200001e8! Aborting.
===== must have access denied on k_sem 0x20001fbc
PASS - test_dyn_thread_perms
===================================================================
Test suite thread_dynamic failed.
===================================================================
PROJECT EXECUTION FAILED

ioannisg, 2019-02-08:
Looks like a stack overflow - need to see why.

ioannisg, 2019-02-08:
The test fails on NXP_MPU (frdm_k64f), on ARMv8-M MPU (nrf9160_pca10090) - does not fail on ARMv7-M MPU (nrf52840_pca10056).

It turns out we need CONFIG_TEST_EXTRA_STACKSIZE be defined, to something other than 0; seems 256 is, not, not enough for te dynamic_thread.

I am trying to see why it _is_ enough for nRF52 :) Maybe something with power-of-two alignment in ARMv7-M MPU that makes the configured stack sufficient, @agross-linaro @andrewboie ?

Otherwise, to me a **quick** fix is CONFIG_TEST_EXTRA_STACKSIZE=256, or somrething similar.

ioannisg, 2019-02-08:
BTW, thanks for reporting this, @arun1joshi

andrewboie, 2019-02-08:
I think we should kick the stack size up to 512 for that test and call it a day, not sure how 256 was even initially arrived at

andrewboie, 2019-02-08:
@ioannisg This reveals another problem: stack overflows are not being reported as such, at least on the NXP MPU

agross-linaro, 2019-02-08:
yeah i am taking a look at this one right now

andrewboie, 2019-02-08:
#7706 strikes again

ioannisg, 2019-02-08:
@andrewboie that would be my next bug fix.

ioannisg, 2019-02-08:
probably on Monday;)

agross-linaro, 2019-02-08:
@ioannisg So the void _arch_configure_dynamic_mpu_regions(struct k_thread *thread) pushes a ton of information on the stack (236 bytes on k64f) for the:
struct k_mem_partition dynamic_regions[_MAX_DYNAMIC_MPU_REGIONS_NUM];

Is this really necessary or is there a way to do this on a single region at a time?

ioannisg, 2019-02-12:
@arun1joshi this test does not fail anymore in master, after we placed the re-programming of the mPU in the priv stack.

@agross-linaro, @galak , still we need to fix the heavy stack usage thing.
But I think this bug report issue can be closed.

agross-linaro, 2019-02-12:
I'm ok with that.

ioannisg, 2019-02-14:
Closing this, then.

closed 2019-02-14 by ioannisg