zephyrproject-rtos/zephyr #13170: Porting guide for advanced board (multi CPU SoC)

r8d8, 2019-02-08:
**Is your enhancement proposal related to a problem? Please describe.**
Started to work with new triplemode SoC(BLE 5.0, WiFi,  802.15.4) board from Qualcomm using [QCA4020 board](https://developer.qualcomm.com/hardware/qca4020-qca4024)
It utilizes 3 CPUs: Cortex-M4(application), Cortex-M0(BLE, 802.15.4), Xtensa7 (WiFi).
Have read "Porting" doc multiple time, but still strugling to port zephyr fro this board. The ain issue for me is deeply nested file structure (Device Tree files) and Kconfig.

**Describe the solution you'd like**
Would be nice to have some tutorial for advanced board (like mine with couple of CPU), that guides user step-by-step through all process of porting

henrikbrixandersen, 2022-11-02:
CC: @mbolivar-nordic

mbolivar-nordic, 2022-11-02:
Yeah, this fits into https://github.com/zephyrproject-rtos/zephyr/issues/51833 I think. We would need to solve this problem for documenting the combination of https://github.com/zephyrproject-rtos/zephyr/issues/51830 and https://github.com/zephyrproject-rtos/zephyr/issues/51831. @r8d8 in short your use case is known and especially since you have different arches, you'll need multiple board directories with different DTS files inside. We don't have a great solution right now.

I'm going to close this issue since the other issues I've linked should address your use case, as documentation for the new enhancements is assumed. Please feel free to comment in those issues if there are requirements you're missing. And please reopen if there's something I missed.

mbolivar-nordic, 2022-11-02:
Sorry, I am not sure why github assumed 'completed' was the status. I meant to close as 'duplicate'.

closed 2022-11-02 by mbolivar-nordic