zephyrproject-rtos/zephyr #13192: silabs flash_gecko driver warning (will fail in CI)

galak, 2019-02-08:
Reproduce by:

```
cd samples/drivers/flash_shell/
mkdir build
cd build/
cmake -DBOARD=efr32mg_sltb004a ..
make
```

output is:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       45740 B         1 MB      4.36%
            SRAM:       11708 B       256 KB      4.47%
        IDT_LIST:          88 B         2 KB      4.30%
/opt/zephyr-sdk-0.9.5/sysroots/x86_64-pokysdk-linux/usr/libexec/arm-zephyr-eabi/gcc/arm-zephyr-eabi/6.2.0/real-ld: warning: orphan section `.ram' from `libzephyr.a(em_msc.c.obj)' being placed in section `.ram'.
```

mnkp, 2019-02-09:
I'll need a bit of help here. Actually, I've already mentioned the issue in #9945

> However, there is still the issue with RAM functions. This driver relies on MSC library module provided by SiLabs which is located in ext/ folder. The module tries to place MSC_WriteWord and MSC_ErasePage functions in .ram section. This section is however missing from Zephyr linker script. That results in a warning about an orphan section and the two functions are placed in flash, not RAM.
> 
> During flash write or erase operation the flash is not accessible to the CPU, i.e. if CPU attempts to read an instruction from flash it will stall. If MSC_WriteWord and MSC_ErasePage functions are placed in flash the CPU will stall immediately, if they are placed in RAM we open a possibility to place other critical ISRs in RAM which then can run even if flash is temporarily inaccessible. This is not straightforward as one would need to ensure that no other kernel, ISR code placed in flash is invoked in the meantime but at least we don't close such a possibility. I didn't test it but I expect that due to design of AHB bus a DMA transfer between peripheral and RAM would still be possible even if CPU is stalled.
> 
> A page erase operation can take up to 40 ms so it's a pretty long time.
> 
> I see following solutions:
> 
> 1. Modify MSC module code placed in ext/ to use Zephyr's macro to place the function code in RAM.
> 2. Add .ram section to Zephyr's linker script.
> 3. Modify MSC module code placed in ext/ not to place the two functions in RAM.
> 4. Implement the two functions natively in this driver so we don't need to touch ext/ folder neither the Zephyr's linker script.

galak, 2019-02-09:
Adding @andrewboie since I feel like I saw a commit from him about marking functions to be placed in ram related to MPU/memory mgmt work.

So I wondering if we can use that same functionality.

@mnkp do you know how the silabs code is placing or defining .ram as the section for the function?

mnkp, 2019-02-11:
To place the function in RAM silabs code is decorating it with `__attribute__ ((section(".ram")))`. The linker script ensures that section .ram is actually located in RAM. That's a standard technique Zephyr is also using.

However, looking at the latest ARM linker script or `include/linker/section_tags.h`, Zephyr seems to be missing a dedicated section for the RAM functions.

galak, 2019-02-11:
I see no issue with adding that to the linker sections.

Take a look at https://github.com/zephyrproject-rtos/zephyr/pull/11228

PR #13692 (2019-02-23): ext: silabs: rename ramfunc section to match Zephyr's
Functions which have to be located in RAM are placed by SiLabs library code in section '.ram'. Corresponding section used by Zephyr is called '.ramfunc'. This patch renames SiLabs section name to match Zephyr's.

This solution was chosen over adding '.ram' function to Zephyr linker scripts not to pollute them. However, this patch needs to be reapplied every time SiLabs library version is updated.

Fixes #13192
commit abc394bd (2019-02-23): ext: silabs: rename ramfunc section to match Zephyr's

Functions which have to be located in RAM are placed by SiLabs library
code in section '.ram'. Corresponding section used by Zephyr is called
'.ramfunc'. This patch renames SiLabs naming to match Zephyr's.

This solution was chosen over adding '.ram' function to Zephyr linker
scripts not to pollute them. However, this patch needs to be reapplied
every time SiLabs library version is updated.

Fixes #13192
closed 2019-02-23 by nashif