zephyrproject-rtos/zephyr #13194: soc/arm/nordic_nrf/nrf52/soc_power.h warning spew when CONFIG_SYS_POWER_MANAGEMENT=n

mbolivar, 2019-02-08:
**Describe the bug**

The above mentioned soc_power.h declares function prototypes which use `enum power_states`, provided by power.h.

However, power.h only defines this enum if CONFIG_SYS_POWER_MANAGEMENT=y. Otherwise, the enum is undefined and leads to the following warnings emitted when building any .c file that includes soc_power.h, directly or indirectly:

```
/home/mbolivar/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52/soc_power.h:29:31: warning: ‘enum power_states’ declared inside parameter list will not be visible outside of this definition or declaration
 void sys_set_power_state(enum power_states state);
                               ^~~~~~~~~~~~
/home/mbolivar/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52/soc_power.h:34:36: warning: ‘enum power_states’ declared inside parameter list will not be visible outside of this definition or declaration
 void sys_power_state_post_ops(enum power_states state);
                                    ^~~~~~~~~~~~
```

This is a *lot* of files, so it would be nice to avoid this issue when power management is disabled by providing alternate prototypes that don't cause warnings.

**To Reproduce**
Steps to reproduce the behavior:
1. Build any application which has CONFIG_SYS_POWER_MANAGEMENT=n. (I used the Zephyr SDK 0.9.5 in case the gcc warning output is version-specific)
2. Look at warnings in build output

**Expected behavior**
No warnings

**Impact**
Lots and lots of warnings :)

**Screenshots or console output**
See above

**Environment (please complete the following information):**
 - OS: all
 - Toolchain: any recent-ish gcc should do
 - Commit SHA: f2ef52f12240cd891aa471e5984fd22a3a1f5e93

**Additional context**
N/A

mbolivar, 2019-02-08:
@carlescufi not sure who this should go to; could you please add the right person?

carlescufi, 2019-02-08:
@anangl @pizi-nordic could you please provide a patch for this?

closed 2019-02-12 by carlescufi