zephyrproject-rtos/zephyr #13197: LwM2M: support Connectivity Monitoring object (object id 4)

mike-scott, 2019-02-08:
**Is your enhancement proposal related to a problem? Please describe.**
Object 4 Connectivity Monitoring is Part of the LwM2M Tech Spec 1.0 and allows connected devices the ability to share statistics such as RSSI, bandwidth usage and other data.  Currently, Zephyr doesn't support this.

**Describe the solution you'd like**
Implement Connectivity Monitoring object in Zephyr.  Useful for modem-based, 802.15.4, bluetooth and wireless connections.

mike-scott, 2019-07-31:
This is implemented in https://github.com/zephyrproject-rtos/zephyr/pull/17710

ioannisg, 2019-08-07:
> 
> 
> This is implemented in #17710

Can the issue be closed, then, @mike-scott ?

closed 2019-08-24 by mike-scott