zephyrproject-rtos/zephyr #13209: FATAL ERROR: unknown key "posixpath" in format string "{posixpath}"

galak, 2019-02-09:
Commit 7e9d1bdda4a50dc66b6429739ff487a5bc959476 has introduced the following message on a build:

`FATAL ERROR: unknown key "posixpath" in format string "{posixpath}"`

I'm guessing this is from west version not being up to date, but its unclear to me how this is suppose to work.  Do we need a minimum version of west installed and checked for?

galak, 2019-02-09:
```
[galak@spiff build]$ west --version
West bootstrapper version: v0.5.1 (/home/galak/.local/lib/python3.7/site-packages/west/_bootstrap)
West repository version: v0.5.1-9-g3ddfd0d (/home/galak/git/.west/west)
```

nashif, 2019-02-09:
just run `west update` and this will go away

galak, 2019-02-09:
I get that, but it seems we should be doing a version check on west if we need a particular version.

tejlmand, 2019-02-11:
As @nashif mentions, running `west update` will fix this, and this is actually expected workflow.

After a lot of talks, with input from many people, it was decided to keep some similarities to git submodules. This means the `west update` can be compared to `git submodule update`.
That is, after pulling a new revision of your main repository with `git`, then you should run `west update`, just as if Zephyr had been using submodules.

The file: https://github.com/zephyrproject-rtos/zephyr/blob/master/west.yml
does contains revisions for both `west` itself (currently `master`), as well as any Zephyr related project (net-tools).

But the only way to get to the correct revision of any project, is by running `west update`

So I think the only way forward is to be better at informing people that `west update` must be run after a `git 'fetch + merge/rebase'` or `git pull` on the Zephyr repo. 

There were talks about combining the `git` operation on Zephyr repo into `west` commands, but as people may have vary different workflows in this repo, it was decided to keep the `git` operation on the manifest repo seperated from `west` commands.

So not sure if this is a bug, as the workflow is expected to be that you should do a `west update` after updating Zephyr repo.
Any sub-project could have changed.
Currently only `net-tools` and `west`, but more projects will follow in future.

SebastianBoe, 2019-02-11:
`after updating Zephyr repo`

Just to be clear, this doesn't just apply to updating the repo, but also downgrading, and therefore git checkout of any zephyr branch, right?

tejlmand, 2019-02-11:
> Just to be clear, this doesn't just apply to updating the repo, but also downgrading, and therefore git checkout of any zephyr branch, right?

Right.
When saying updating, it is updating / checkout to any revision (SHA/pull/branch).
Whether an update is actually an upgrade / downgrade doesn't matter (of course going to a revision prior-west, ie. without `west.yml` will not need `west update`)

galak, 2019-02-11:
> Right.
> When saying updating, it is updating / checkout to any revision (SHA/pull/branch).
> Whether an update is actually an upgrade / downgrade doesn't matter (of course going to a revision prior-west, ie. without `west.yml` will not need `west update`)

1. Can we have the error message report back to do an 'west update'. 

2. Why not have a west version check?

mbolivar, 2019-02-11:
> * Can we have the error message report back to do an 'west update'.
> * Why not have a west version check?

We are going to do a minimum version check using `git merge-base --is-ancestor MANIFEST_REVISION CURRENT_WEST_REVISION`. That will emit a warning if your west is not a descendant of the manifest revision (as part of this, we're going to start putting a SHA as the west revision in the manifest, west.yml).

This won't be able to detect if your west is too new. That will require moving all the built-in commands into what gets installed via PyPI, deleting the clone of west in the installation directory .west/west, and using upper/lower bounds on the current west semantic version. This is planned for v1.15.

erwango, 2019-02-12:
This issue impacts everyone. Waiting for a solution that make the issue disappears, I'm pinning this issue so people avoid losing too much time digging into this.

carlescufi, 2019-02-12:
This requires documentation and education of the user-base first and foremost IMHO. We need to communicate clearly that `west update` is required often, especially in this initial phase where we are ironing out the quirks and bugs left in west. Later down the line this will not be as critical.

galak, 2019-02-12:
> This requires documentation and education of the user-base first and foremost IMHO. We need to communicate clearly that `west update` is required often, especially in this initial phase where we are ironing out the quirks and bugs left in west. Later down the line this will not be as critical.

I still haven't heard or understood, why the check of west version or SHA or whatever isn't happening.  I've got no issue doing `west update`.  Just we need to say that it needs to occur (at least for west itself dependancies).  I get the need for it once we start having other repo's.

carlescufi, 2019-02-12:
>I still haven't heard or understood, why the check of west version or SHA or whatever isn't happening. I've got no issue doing `west update`. Just we need to say that it needs to occur (at least for west itself dependancies). I get the need for it once we start having other repo's.

We will add the check, those are orthogonal. But even with the check you can forget to run `west update` and west won't be able to warn you, because even if we warn the user that the checked out version for a particular repo is not a direct descendant of the one in the manifest, this will not prevent you having a west that is *too recent*. That said, @mbolivar (who already mentioned this above) will send a PR to add the SHA check.

galak, 2019-02-12:
> We will add the check, those are orthogonal. But even with the check you can forget to run `west update` and west won't be able to warn you, because even if we warn the user that the checked out [...]

I feel like there is some confusion, I'm specifically talking about west as a tool of the build system and having a version check for it.  Not about other repo's.  In the same way we check if the DTC version you point at is valid, we should check the west version and report an error that says something like 'do a west update'.  I get that there is repo mgmt that requires 'west update', but I view that as a different issue.

commit 6929c374 (2019-03-12): cmake: add west version checking

Parse the west version output and make sure both bootstrapper and
installation wests are up to date. This will still work after the two
are combined into what gets installed via PyPI.

Fixes: #13209
closed 2019-03-12 by andrewboie