zephyrproject-rtos/zephyr #13211: net/sockets: send/sendto broken

aurel32, 2019-02-09:
**Describe the bug**

The send/sendto functions used in a TCP connection always return an error.

This has been introduced by commit 083470a14a52234f494ad2a302497d309657ac79.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `samples/net/sockets/dumb_http_server`
2. Try to download a file from the board using for example `wget` or `curl`
3. The download fails

**Expected behavior**
Download from the board is successful.

**Impact**
The IP stack is unusable.

**Screenshots or console output**
Logs from Zephyr:
```
[00:00:00.000,040] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,220] <inf> eth_sam: MAC: fc:c2:3d:12:e9:21
[00:00:00.001,312] <inf> eth_sam: Queue 0 activated
[00:00:00.001,315] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.140,015] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
***** Booting Zephyr OS zephyr-v1.13.0-3876-g083470a14a *****
[00:00:02.820,016] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:02.825,337] <inf> net_config: Initializing network
[00:00:02.825,360] <inf> net_config: IPv4 address: 192.0.2.1
Single-threaded dumb HTTP server waits for a connection on port 8080...
Connection #0 from 192.0.2.2
Error sending data to peer
Connection from  closed
```

Logs from the host:

```
$ wget 192.0.2.1:8080
--2019-02-09 16:28:41--  http://192.0.2.1:8080/
Connecting to 192.0.2.1:8080... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2019-02-09 16:28:42--  (try: 2)  http://192.0.2.1:8080/
Connecting to 192.0.2.1:8080... connected.
HTTP request sent, awaiting response... No data received.
Retrying.

--2019-02-09 16:28:44--  (try: 3)  http://192.0.2.1:8080/
Connecting to 192.0.2.1:8080... connected.
HTTP request sent, awaiting response... No data received.
Retrying.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Debian toolchain
 - Commit SHA: any commit more recent than 083470a14a52234f494ad2a302497d309657ac79

aurel32, 2019-02-09:
The problem happens when `send` or `send_to` are use on buffer bigger than the MTU. In that case  `net_pkt_cursor_operate` returns with -ENOBUFS.

pfalcon, 2019-02-10:
Confirmed that samples/net/sockets/echo works (for trivial data typed in).

Confirmed that samples/net/sockets/dumb_http_server has the behavior described by @aurel32 .

tbursztyka, 2019-02-11:
Will take a look.
Downgraded the priority to medium, the ip stack is not unusable (not entirely)

tbursztyka, 2019-02-11:
Found the issue, forgot to adjust the given length to actual buffer space in net_pkt. Easy fix. Will send a patch soon.

aurel32, 2019-02-11:
> Found the issue, forgot to adjust the given length to actual buffer space in net_pkt. Easy fix. Will send a patch soon.

Thanks for the additional patch, I confirm that it's possible to send big TCP payload with the GMAC driver again.

PR #13246 (2019-02-12): net context: write only the maximum payload net_pkt can hold
Attempt to fix #13211 

@aurel32 , @pfalcon I could not get internet access to qemu. I have an heavy setup on my work laptop (vpn, proxies etc...) and it's probably getting on the way. Could you check?
commit 16edbbad (2019-02-12): net/context: Update length according to available buffer for payload

UDP was the only one doing the right thing. Let's generalize it
relevantly.

Fixes #13211
closed 2019-02-12 by nashif