zephyrproject-rtos/zephyr #13212: RFC: Requirements for implementations of high(er)-level networking protocols in Zephyr

pfalcon, 2019-02-09:
## Introduction

This RFC proposes a set of requirements and guidelines to design API structure, contracts, and behavior for application-level network protocols in Zephyr.

### Problem description

Zephyr has some history (arguably, dramatic history) of networking in general, and implementations for application-level protocols in particular. Older, net-app based protocol implementation had various reported issues and criticism, and eventually were removed completely together with net-app itself.

Starting with Sockets API as a baseline allows us to design new implementations to address previously known issues, and at the same time, to be consistent and compatible among themselves. We should pursue that opportunity. 

### Summary

Application-level protocols should:

* Not impose arbitrary restrictions on the amount of *data* processed (as an none of general-purpose networking protocols imposes such arbitrary restrictions).
* May impose arbitrary restrictions on *metadata* size, but should try to make those restrictions/limits configurable (at compile time).
* Should follow the general structure and signatures of the underlying API, as a means of design consistency and self-organization.
* Should strive for composabilty (largely enabled by the previous point). 

## Detailed RFC

In this section of the document the target audience is the dev team. Upon
reading this section each engineer should have a rather clear picture of what
needs to be done in order to implement the described feature.

### Proposed change (Detailed)

This section is freeform - you should describe your change in as much detail
as possible. Please also ensure to include any context or background info here.
For example, do we have existing components which can be reused or altered.

By reading this section, each team member should be able to know what exactly
you're planning to change and how.

### Dependencies

Highlight how the change may affect the rest of the project (new components,
modifications in other areas), or other teams/projects.

### Concerns and Unresolved Questions

List any concerns, unknowns, and generally unresolved questions etc.

## Alternatives

List any alternatives considered, and the reasons for choosing this option
over them.

zagor, 2019-02-20:
Some thoughts:

The de-facto standard of HTTP client libraries is libcurl. The libcurl API also fulfils many of the requirements listed here. However, libcurl is not designed with an embedded focus and for example makes liberal use of dynamic memory. It's also optimised for completeness rather than compactness.

I know @bagder is currently tweaking curl to make it possible to build it with more limited feature sets but even so it will probably never be as small as a bespoke solution that "just solves one problem" and disregards the complex world of incompatible internet protocol implementations.

This raises the question of what standard to aim for in Zephyr. I suspect many people will be looking for a very simple and compact solution ("_I just need to download a text file from my backend_"), while others want a more comprehensive and bug-compatible one ("_we have to be able to interact with any public server_"). Neither user will be happy with the solution preferred by the other. And it is not easy to supply both variants in a single package.

chinli, 2019-03-28:
@pfalcon how's it going about the http lib? what the schedule of it?

andrewboie, 2019-03-28:
Higher-level protocols need to be compatible with user mode.

chinli, 2019-03-28:
@pfalcon @andrewboie Rightnow, is there any alternative solution?

zagor, 2019-04-01:
FYI we have written a simple and fully static (no malloc) http server in my current project at Xylem, using the Zephyr http-parser, tls and sockets. I have secured approval to upstream this and will create a pull request for initial discussions soon.

rveerama1, 2019-04-02:
@zagor we have Zephyr Monthly Networking forum today. This would be a good platform to discuss Networking topic.

I am posting meeting details here. Please try to join this meeting.
Please check meeting time as per your location.

https://www.zephyrproject.org/developers/meetings/

Zephyr Networking Forum (monthly recurrence 2019)
Tuesday, 2 April⋅6:00 – 7:00pm (Finnish time).
Monthly on the first Tuesday
https://zoom.us/j/720590301
Description:Live Agenda/Minutes: https://docs.google.com/document/d/1qFsOpvbyLzhflJbbv4Vl__497pKHDoUCy9hjAveyCX0/edit?usp=sharing

Shared Folder:
https://drive.google.com/drive/folders/1j6d0FLeOjiMil1Ellb59AsfHdzuWdAAc?usp=sharing

-----

Join from PC, Mac, Linux, iOS or Android: https://zoom.us/j/720590301

Or iPhone one-tap :
    US: +16699006833,,720590301# or +16465588656,,720590301# 
Or Telephone:
    Dial(for higher quality, dial a number based on your current location): 
        US: +1 669 900 6833 or +1 646 558 8656 or +1 877 369 0926 (Toll Free) or +1 855 880 1246 (Toll Free)
    Meeting ID: 720 590 301
    International numbers available: https://zoom.us/zoomconference?m=HfPX4_ydWipHdthccFN1H_TJg-7jDSyH

pfalcon, 2019-04-25:
@rlubos: I see that somehow you weren't cc:ed here, even though this was posted as generalization of our discussion re: MQTT API. Doing this now.

closed 2024-02-09 by nashif