zephyrproject-rtos/zephyr #13220: qemu_x86_64 build failures

galak, 2019-02-09:
The following tests fail to build:

```
qemu_x86_64               tests/lib/mem_alloc/libraries.libc.newlib          FAILED: qemu_x86_64               tests/misc/test_build/test_newlib                  FAILED: qemu_x86_64               samples/net/sockets/http_get/test                  FAILED: 
qemu_x86_64               samples/net/sockets/dumb_http_server/test          FAILED: qemu_x86_64               samples/net/sockets/big_http_download/test         FAILED:
```

andyross, 2019-02-10:
With the exception of a few (passing) subtests, those all require newlib, which a host gcc obviously doesn't have.  They are all whitelisted in sanitycheck, so I don't know where the failures you're seeing are coming from:

```
$ scripts/sanitycheck -v -p qemu_x86_64 -T tests/lib/mem_alloc  -T tests/misc/test_build -T samples/net/sockets/http_get -T samples/net/sockets/dumb_http_server -T samples/net/sockets/big_http_download
JOBS: 24
Cleaning output directory /home/andy/zephyr/sanity-out
Reading test case configuration files under /home/andy/zephyr/tests/lib/mem_alloc...
Reading test case configuration files under /home/andy/zephyr/tests/misc/test_build...
Reading test case configuration files under /home/andy/zephyr/samples/net/sockets/http_get...
Reading test case configuration files under /home/andy/zephyr/samples/net/sockets/dumb_http_server...
Reading test case configuration files under /home/andy/zephyr/samples/net/sockets/big_http_download...
Reading platform configuration files under /home/andy/zephyr/boards...
Reading platform configuration files under /home/andy/zephyr/scripts/sanity_chk/boards...
Building testcase defconfigs...
3 tests selected, 1567 tests discarded due to filters
1/3 qemu_x86_64               tests/misc/test_build/test_debug                   PASSED
2/3 qemu_x86_64               tests/misc/test_build/test_utf8_in_kconfig_values  PASSED
3/3 qemu_x86_64               tests/lib/mem_alloc/libraries.libc.minimal         PASSED
3 of 3 tests passed with 0 warnings in 4 seconds
```

nashif, 2019-02-10:
my bug.

nashif, 2019-02-10:
hmm, looks like for x86_64 we use the toolchain variant "zephyr" and just revert to host tools for x86_64 which is kind of a workaround, the side effect is that we take some assumptions valid for the zephyr variant.
With the new change to sanitycheck to allow building for multiple variants, I think we can remove this workaround and declare this as supported by host variant only.

PR #13227 (2019-02-10): X86_64 should use host compiler and the host-gcc definition
We had this working with the Zephyr variant and forcing host compiler from gcc
compiler definition which is used for cross toolchains.

Fixes #13220
closed 2019-02-10 by galak