zephyrproject-rtos/zephyr #13236: Failure tests/kernel/gen_isr_table on some stm32 platforms

galak, 2019-02-10:
Seeing failures on the following platforms:

```
olimexino_stm32           tests/kernel/gen_isr_table/arch.interrupt          FAILED: build_error
nucleo_f103rb             tests/kernel/gen_isr_table/arch.interrupt          FAILED: build_error
stm32_min_dev             tests/kernel/gen_isr_table/arch.interrupt          FAILED: build_error
```

erwango, 2019-02-11:
So it appears generated table is as follows:
```
gen_isr_tables.py: multiple registrations at table_index 40 for irq 40 (0x28)
gen_isr_tables.py: (43, 0)
gen_isr_tables.py: Configured interrupt routing
gen_isr_tables.py: handler    irq flags param
gen_isr_tables.py: --------------------------
gen_isr_tables.py: 0x8000111  39  0     0xca55e77e
gen_isr_tables.py: 0x80000ed  40  0     0xb01dface
gen_isr_tables.py: 0x80001a9  41  1     0x0
gen_isr_tables.py: 0x800017d  42  1     0x0
gen_isr_tables.py: 0x8000c2d  40  0     0x20000f74
gen_isr_tables.py: 0x8000c23  23  0     0x20000f74
gen_isr_tables.py: 0x8000c19  10  0     0x20000f74
gen_isr_tables.py: 0x8000c0f  9   0     0x20000f74
gen_isr_tables.py: 0x8000c05  8   0     0x20000f74
gen_isr_tables.py: 0x8000bfb  7   0     0x20000f74
gen_isr_tables.py: 0x8000bf1  6   0     0x20000f74
gen_isr_tables.py: offset is 0
gen_isr_tables.py: num_vectors is 43

```

All irq generated with touring param `0x20000f74` are coming from external interrupt driver.
Among which irq n 40 matches the irq EXTI15_10_IRQn. So it appears to be correct.

Supernumerary irq 40 is the following one:
```
gen_isr_tables.py: 0x80000ed  40  0     0xb01dface
```
It appears to be made up by the test:
```
#define ISR3_OFFSET	2
#define ISR3_ARG	0xb01dface
#define IRQ_LINE(offset)	(CONFIG_NUM_IRQS - ((offset) + 1))
[...]
	IRQ_CONNECT(IRQ_LINE(ISR3_OFFSET), 1, isr3, ISR3_ARG, 0);
	irq_enable(IRQ_LINE(ISR3_OFFSET));
```

Seems that tests installs a test IRQ at and arbitrary index in the table, but no luck, for that target this index is actually already in use. Since choice of the ISR and configuration of the existing activated ISR are both arbitrary changing the index of ISR3 would only move the problem to another target.

One way to avoid this would be to disable all drivers using Kconfig before building.
I don't know if this is compatible with the aim of the test, but I guess this should be ok. 
Though, to do it in a simple was it would require some extra logic in Kconfig like a meta Kconfig symbol (eg ZEPHYR_DRIVER from which all Kconfig high level drivers symbols would depend (UART, I2C, GPIO, ..).

galak, 2019-02-11:
I realized, and I think for now the best choice is to just exclude the failing platforms from the test.  I'll send a PR for this.

closed 2019-02-14 by galak