zephyrproject-rtos/zephyr #13237: stm32 USB sanitycheck failures with sdk 0.10.0-beta2

galak, 2019-02-10:
Seeing a number of sanitycheck failures due the following warning when building with the newer gcc in sdk 0.10.0-beta:

```
/home/galak/git/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c: In function 'USB_WritePacket':
/home/galak/git/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:886:7: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'unsigned int *'} [-Werror=attributes]
       USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
       ^~~~~~~~~~
/home/galak/git/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c: In function 'USB_ReadPacket':
/home/galak/git/zephyr/ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:912:5: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'unsigned int *'} [-Werror=attributes]
     *(__packed uint32_t *)dest = USBx_DFIFO(0U);
     ^
cc1: all warnings being treated as errors

```

Reproduce:

```
./scripts/sanitycheck -T samples/subsys/usb/cdc_acm -l
```

galak, 2019-02-10:
Not sure what the proper workaround for this.  Not clear to me why the cast to (__packed uint32_t *) is even there.

In previous similar issue, I fixed this with UNALIGNED_GET() see commit 93d0f8c58dbe973f95f7e8b4f7a23dd4e97dd45b.

Whatever fix will probably need to be applied to all these cases:
```
ext/hal/st/stm32cube/stm32f1xx/drivers/src/stm32f1xx_ll_usb.c:    USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
ext/hal/st/stm32cube/stm32f1xx/drivers/src/stm32f1xx_ll_usb.c:    *(__packed uint32_t *)dest = USBx_DFIFO(0);
ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c:      USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c:    *(__packed uint32_t *)dest = USBx_DFIFO(0U);
ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:      USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
ext/hal/st/stm32cube/stm32f4xx/drivers/src/stm32f4xx_ll_usb.c:    *(__packed uint32_t *)dest = USBx_DFIFO(0U);
ext/hal/st/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_usb.c:      USBx_DFIFO((uint32_t)ch_ep_num) = *((__packed uint32_t *)pSrc);
ext/hal/st/stm32cube/stm32f7xx/drivers/src/stm32f7xx_ll_usb.c:    *(__packed uint32_t *)pDest = USBx_DFIFO(0U);
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_ll_usb.c:    USBx_DFIFO((uint32_t)ch_ep_num) = *((__packed uint32_t *)pSrc);
ext/hal/st/stm32cube/stm32l4xx/drivers/src/stm32l4xx_ll_usb.c:    *(__packed uint32_t *)pDest = USBx_DFIFO(0U);
```

erwango, 2019-02-11:
What about using CMSIS defines ?
```
__UNALIGNED_UINT32_READ
__UNALIGNED_UINT32_WRITE
```
Downloading sdk_ng, so I haven't been able to test.
Though, since available in cmsis pack should be easier to propagate in Cube releases

galak, 2019-02-11:
This makes the warnings go away for me, please look to see its correct and I think you can make the change for the other driver instances:

```
diff --git a/ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c b/ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c
index a0e37e8afd..e9872e3d2d 100644
--- a/ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c
+++ b/ext/hal/st/stm32cube/stm32f2xx/drivers/src/stm32f2xx_ll_usb.c
@@ -768,7 +768,7 @@ HAL_StatusTypeDef USB_WritePacket(USB_OTG_GlobalTypeDef *USBx, uint8_t *src, uin
     count32b =  (len + 3U) / 4U;
     for (i = 0U; i < count32b; i++, src += 4U)
     {
-      USBx_DFIFO(ch_ep_num) = *((__packed uint32_t *)src);
+      USBx_DFIFO(ch_ep_num) = __UNALIGNED_UINT32_READ(src);
     }
   }
   return HAL_OK;
@@ -794,7 +794,7 @@ void *USB_ReadPacket(USB_OTG_GlobalTypeDef *USBx, uint8_t *dest, uint16_t len)
 
   for ( i = 0U; i < count32b; i++, dest += 4U )
   {
-    *(__packed uint32_t *)dest = USBx_DFIFO(0U);
+    __UNALIGNED_UINT32_WRITE(dest, USBx_DFIFO(0U));
 
   }
   return ((void *)dest);
```

erwango, 2019-02-11:
Tested ok on nucleo_f429zi.
Will submit internal ST issues and make fixes in cube packages

closed 2019-02-11 by galak

marc-hb, 2019-02-11:
As a coincidence I just asked and got confirmation that packed is poorly defined: https://groups.google.com/d/msg/x86-64-abi/IT0wgB6Fd5g/Vd-hg7YVAwAJ
Just FYI.

PR #13275 (2019-02-11): ext/hal: stm32: Use of (__packed uint32_t *) produces warnings
Using Zephyr SDK 0.10.0-rc2, GNUCC 8.2.0 is used and (__packed uint32_t *) are now generating warnings..
Replace with CMSIS macros __UNALIGNED_UINT32_READ and __UNALIGNED_UINT32_WRITE

Fixes #13237
commit d30975b0 (2019-02-11): ext/hal: stm32l4xx: Use of (__packed uint32_t *) produces warnings

Using Zephyr SDK 0.10.0-rc2, GNUCC 8.2.0 is used and
(__packed uint32_t *) are now generating warnings..
Replace with CMSIS macros __UNALIGNED_UINT32_READ and
__UNALIGNED_UINT32_WRITE

ST internal reference: 61328

Fixes #13237