zephyrproject-rtos/zephyr #13243: DT: error in generated_dts_board_fixups.h for board: frmd_k64f

jarz-nordic, 2019-02-11:
It is not possible to build application:
~/zephyr/zephyr/tests/net/all
for frdm_k64f board.

1. Go to ~/zephyr/zephyr/tests/net/all
2. mkdir b && cd b
3. cmake -GNinja -DBOARD=frdm_k64f ..
4. ninja
5. See error

**Expected behavior**
Project will compile

**Impact**
It is not possible to build this test for frdm_k64f board. In my opinion it is serious.

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - Commit SHA or Version used: acaae579ccb4615a06900bdd618fa34a056b6eac (the newest master)

**Screenshots or console output**

[image]

**Additional context**
I can build it without any error for nrf52840_pca10056 board.

galak, 2019-02-11:
The issue is that the build isn't valid on frdm-k64f as defined by default.  It enables bluetooth, but there is no bluetooth controller.  BT_UART is enabled and expects that `zephyr,bt-uart` is defined in the dts, but since the based frdm-k64f doesn't have bt, that isn't set.  Thus the build errors.

MaureenHelm, 2019-02-11:
Is this purely a build test or is it supposed to do something at runtime? If it's just a build test, then you can build it like this:
```
$ cmake -DBOARD=frdm_k64f -DSHIELD=frdm_kw41z ..
$ make
```
It won't work at runtime because there pin conflict between Ethernet and BT UART.

agansari, 2019-02-12:
This issue will arise on any board without Bluetooth on board/soc that requires an external shield, not just K64F.
To fix this we can add something like the following:

`diff --git a/tests/net/all/CMakeLists.txt b/tests/net/all/CMakeLists.txt
index 4491e40bd9..6a5aaddde3 100644
--- a/tests/net/all/CMakeLists.txt
+++ b/tests/net/all/CMakeLists.txt
@@ -1,4 +1,9 @@
 cmake_minimum_required(VERSION 3.13.1)
+
+if ((BOARD STREQUAL "frdm_k64f") OR (BOARD STREQUAL "mimxrt1050_evk"))
+    set(SHIELD frdm_kw41z)
+endif()
+
 include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)
 project(all)
`

But as @MaureenHelm asked, is this a build test?
If yes, then the change above will fix it.
If no, then you need a particular HW setup described here: https://docs.zephyrproject.org/latest/boards/shields/frdm_kw41z/doc/index.html

I expect boards without BT but have arduino-i2c dts nodes to request x_nucleo* shields.

nashif, 2019-02-19:
This is a build test only, the testcase.yaml says it:

```  # Only used for compile testing so it is enough to use one platform here.
  platform_whitelist: qemu_x86```

closed 2019-02-19 by nashif