zephyrproject-rtos/zephyr #13251: samples/net/sockets/echo_server no longer working on frdm_k64f

i8080, 2019-02-11:
echo_server project compiles but crashes on frdm_k64f

1. cmake -GNinja -DBOARD=frdm_k64f ..
2. ninja flash

console output is:

```
***** Spurious interupt (IRQ  *****
***** Hardware exception *****
Cuent thread ID = 0x20001cb0
Fault instruction address = 0x2000aff0
Fatal fault in ISR! Spinning...
```

maybe this is related to #13243 ?

tbursztyka, 2019-02-11:
and PR #13250 seems to fix it

i8080, 2019-02-11:
unfortunately PR #13250 does not fix it, but changes to error to:

```
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x20002bd0
Faulting instruction address = 0x17b08
Fatal fault in ISR! Spinning...
```

pfalcon, 2019-02-11:
Confirmed, also tried samples/net/dns_resolve just to be sure that's not a particular sample, with the same output (crash addresses differ of course).

pfalcon, 2019-02-11:
Confirming that https://github.com/zephyrproject-rtos/zephyr/pull/13250 seems to fix the issue for me.

pfalcon, 2019-02-11:
However, ping doesn't work for me after that, so something is still not right.

pfalcon, 2019-02-11:
Not that dead networking allows to keyboard-fuzz this better (working resolution happens too fast otherwise). For example, what happens if we issue new "net dns query" while previous is still active? Can't crash the stupid thing, it seems:

~~~
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Query for 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' sent.

dns: Timeout while resolving name.
uart:~$ 
uart:~$ 
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Query for 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' sent.
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)

dns: Timeout while resolving name.
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Query for 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' sent.
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-11)

dns: Timeout while resolving name.
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-2)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-2)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-2)
uart:~$ net dns query AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA A
IPv4 address type
Cannot resolve 'AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA' (-2)
~~~

pfalcon, 2019-02-11:
Pings work after retry.

i8080, 2019-02-11:
ping and UDP echo works with #13250 if CONFIG_NET_TCP is disabled

@pfalcon do you have CONFIG_NET_TCP disabled?

agansari, 2019-02-11:
> Pings work after retry.

With #13260 and #13250 both patches.

pfalcon, 2019-02-11:
> @pfalcon do you have CONFIG_NET_TCP disabled?

Nope, I try to use pristine samples whenever possible (first I had some WIP code in my tree, and that could explain the issues I initially saw). I used samples/net/sockets/echo

chtavares592, 2019-02-11:
My application is a coap-client, and when I start the image on the board, the same error occurs.

```
***** Spurious interrupt (IRQ  83) *****g to the first image slot
***** Hardware exception *****
Current thread ID = 0x20001f28
Faulting instruction address = 0x20007e08
Fatal fault in ISR! Spinning..
```

i8080, 2019-02-11:
I used samples/net/sockets/echo_server, that only works if CONFIG_NET_TCP disabled.
A short test with samples/net/sockets/echo and #13250 works fine.

pfalcon, 2019-02-11:
@i8080, @chtavares592: Please retry with the latest master of d6a19afb29294d2dc1799ac3e322a46db83f7b07 . If some issue still persists, please provided the exact revision used and steps to reproduce (sorry for asking to do that all over again), but tracing those steps thru multiple comments is errorprone.

chtavares592, 2019-02-11:
@pfalcon it worked for me!!

i8080, 2019-02-11:
samples/net/sockets/echo_server only works if CONFIG_NET_TCP is disabled

how to reproduce:

1. set CONFIG_NET_TCP=n in prj.conf
2. cmake -DBOARD=frdm_k64f
3. ninja flash
4. ping and udp echo are working fine

5. set CONFIG_NET_TCP=y in prj.conf (this is the default value)
6. ninja flash
7. no reaction on the board

 - OS: Ubuntu Linux 18.04
 - Toolchain Zephyr SDK 0.9.5
 - tested with latest master of d6a19af

pfalcon, 2019-02-11:
> 7\. no reaction on the board

Following steps described by @i8080, I can reproduce this. There's indeed absolutely zero output on UART from the board after pressing reset button. Ethernet connection is also not active.

By making following change to `samples/net/sockets/echo_server/prj.conf`, I can get serial output:

~~~
-CONFIG_MAIN_STACK_SIZE=2048
+CONFIG_MAIN_STACK_SIZE=4096
~~~

However, networking doesn't really. Apparently this was the case I was hitting when I reported "ping don't work" above.

By making following change:
~~~
-CONFIG_NET_SHELL=y
+#CONFIG_NET_SHELL=y
~~~

I get networking (pings, TCP conn) to work.

Assigning @rlubos as the maintainer of this sample.

Pinging @jarz-nordic regarding shell surprises.

Pinging @agansari regarding a hunch that IIRC we kinda had a conflict of UART and Ethernet on imxrt1050, could it occur here on frdm_k64f?

jarz-nordic, 2019-02-11:
@pfalcon : In my opinion it is not related to shell. I observed the same error today in example:
`~/zephyr/zephyr/samples/net/vlan` on frdm_k64f.

I have disable shell and this error was still there. When I disabled shell AND logger error was gone.
@nordic-krch : has investigated that. Error occured AFTER logger intialization and before first logged message. In our opinion it is some kind of a stack problem.

pfalcon, 2019-02-11:
> When I disabled shell AND logger error was gone.

Thanks for that info. Well, logger remains the first suspect for me. Only today I disabled (or switched to IMMEDIATE) on multiple occasions, but *today* it didn't make a difference, and here I randomly flipped shell (based on knowing that socket/echo differs from socket/echo_server exactly by shell being enabled in the latter), and it clicked.

nordic-krch, 2019-02-12:
`***** Spurious interrupt (IRQ  83) *****` could that message be of any help? what is IRQ 83 on that platform.

I briefly tried to debug`net/vlan` sample yesterday. Change to `LOG_IMMEDIATE` and had breakpoint on first log message but system was crashing earlier (but after log_core initialization) so even though logger is a suspect it has alibi for now.

pfalcon, 2019-02-12:
> `***** Spurious interrupt (IRQ 83) *****` could that message be of any help? what is IRQ 83 on that platform.

@nordic-krch, that issues was fixed. This ticket contains 2 unrelated issues reports. I was too shy to ask @i8080 to close this and submit a new one, and it pays off...

Please look at stuff only in and after this comment: 
https://github.com/zephyrproject-rtos/zephyr/issues/13251#issuecomment-462384858

jarz-nordic, 2019-02-12:
Today I've tested: `~/zephyr/zephyr/samples/net/sockets/echo_server`
I have following observation:
1. When I'm generating project with a cmake there is following error: 
> FATAL ERROR: unknown key "posixpath" in format string "{posixpath}".
Despite that project is created and can be build.

2. When I remove either Logger or Shell I do not see interrupt spin error. When both are in prj.conf there is an error:

```
***** BUS FAULT *****
  Imprecise data bus error
***** Hardware exception *****
Current thread ID = 0x20002bd0
Faulting instruction address = 0x17b08
Fatal fault in ISR! Spinning...
```

3. This will be good one! :)
When I changed
`CONFIG_INIT_STACKS=y`
to
`#CONFIG_INIT_STACKS=y`

Example works pefectly with shell AND Logger enabled.

jarz-nordic, 2019-02-12:
Another observation:
I keept all the features as they were **AND** I've addded one more:
`CONFIG_LOG_PRINTK=y` and it solved the problem.

Maybe we have some problem with access to UART. @nordic-krch suggested that:
1. Logger & shell
2. printk
are compeating for UART access at the same time. Maybe one is breaking anothers transfer?
When I've set `CONFIG_LOG_PRINTK=y` all parties accessing UART are synchronized.

i8080, 2019-02-12:
@jarz-nordic @pfalcon 

just opened new issue #13301 and closed this

closed 2019-02-12 by i8080