zephyrproject-rtos/zephyr #13253: tests/drivers/counter/counter_basic_api: nchan  not equal to alarm_cnt

cinlyooi-intel, 2019-02-11:
**Describe the bug**
Test failed complaining about "nchan  not equal to alarm_cnt" when running test_all_channels()

**To Reproduce**
Steps to reproduce the behavior:
0. Modify src/test_counter.c to disable all tests except test_all_channels() in test_main(). This is to avoid "kernel panic" in the test_* ran before test_all_channels() masking this error
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=nrf52_pca10040 -DARCH=arm #nrf52_pca10040
3. ninja
4. ninja run

Replace (2) with
cmake -GNinja -DBOARD=nrf51_pca10028 -DARCH=arm for nrf51_pca10028
cmake -GNinja -DBOARD=nrf52840_pca10056 -DARCH=arm for nrf52840_pca10056

**Screenshots or console output**
For nrf52_pca10040
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-4573-g012d404125 (delayed boot 1000ms) *****
Running test suite test_counter
===================================================================
starting test - test_all_channels
    Assertion failed at /path/to/zephyr.git/tests/drivers/counter/counter_basic_api/src/test_counter.c:354: test_all_channels_instance: (nchan not equal to alarm_cnt)
Expecting alarm callback
FAIL - test_all_channels
===================================================================
Test suite test_counter failed.
===================================================================
[00:00:00.000,000] <dbg> counter_timer.4.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.3.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.2.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.1.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.0.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_rtc.0.init_rtc: Initialized
[00:00:00.000,000] <dbg> counter_rtc.2.init_rtc: Initialized
```

For nrf51_pca10028
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-4573-g012d404125 (delayed boot 1000ms) *****
Running test suite test_counter
===================================================================
starting test - test_all_channels
    Assertion failed at /path/to/tests/zephyr.git/tests/drivers/counter/counter_basic_api/src/test_counter.c:354: test_all_channels_instance: (nchan n
ot equal to alarm_cnt)
Expecting alarm callback
FAIL - test_all_channels
===================================================================
Test suite test_counter failed.
===================================================================
[00:00:00.000,000] <dbg> counter_timer.2.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.1.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.0.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_rtc.0.init_rtc: Initialized
```

For nrf52840_pca10056
```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-4573-g012d404125 (delayed boot 1000ms) *****
Running test suite test_counter
===================================================================
starting test - test_all_channels
    Assertion failed at /path/to/tests/zephyr.git/tests/drivers/counter/counter_basic_api/src/test_counter.c:354: test_all_channels_instance: (nchan 
not equal to alarm_cnt)
Expecting alarm callback
FAIL - test_all_channels
===================================================================
Test suite test_counter failed.
===================================================================
[00:00:00.000,000] <dbg> counter_timer.4.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.3.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.2.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.1.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_timer.0.init_timer: Initialized
[00:00:00.000,000] <dbg> counter_rtc.0.init_rtc: Initialized
[00:00:00.000,000] <dbg> counter_rtc.2.init_rtc: Initialized
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr
 - Commit SHA or Version used: 012d40412568201f975dd0356e1e3fbf9e8aad73

nordic-krch, 2019-03-07:
@cinlyooi-intel can you retest? I couldn't reproduce it.

cinlyooi-intel, 2019-03-08:
@nordic-krch Not observed on 1f74e0b1f75eff63568c1c807f003df7c3b7793b. The test passed.

#13255  had masked this bug because the test case that it covers runs before this test case. As it is fixed by #14148, it should had exposed this problem. It did not. May be it is fixed.

nordic-krch, 2019-03-13:
@cinlyooi-intel i'm closing it then.

closed 2019-03-13 by nordic-krch