zephyrproject-rtos/zephyr #13257: Shell not compatible with c++

rodrigopex, 2019-02-11:
**Describe the bug**
_A clear and concise description of what the bug is.
What have you tried to diagnose or workaround this issue?_

Shell is not compatible with C++.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable C++ and Shell at the prj file;
2. Add code to initialize some shell command using the Shell Macros;
3. mkdir build; cd build
4. cmake -DBOARD=nucleo_f091rc -GNinja ..
5. ninja
6. See error

**Expected behavior**
_A clear and concise description of what you expected to happen._

It must be possible to compile and execute the shell properly with C++.

**Impact**
_What impact does this issue have on your progress (e.g., annoyance, showstopper)_

The code does not compile, so the user would not use the shell on the firmware. Shell is an important feature for testing, interact and updating the code via mcumgr shell backend.

**Screenshots or console output**
_If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue._

Here is the compilation log:
```
../src/main.cpp
In file included from ../src/main.cpp:22:0:
/home/r...t/zephyr/include/shell/shell.h:170:41: error: uninitialized const ‘shell_sub_h’ [-fpermissive]
  static const struct shell_static_entry shell_##name[];                 \
                                         ^
../src/main.cpp:104:1: note: in expansion of macro ‘SHELL_CREATE_STATIC_SUBCMD_SET’
 SHELL_CREATE_STATIC_SUBCMD_SET(sub_h)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/r...t/zephyr/include/shell/shell.h:107:8: note: ‘const struct shell_static_entry’ has no user-provided default constructor
 struct shell_static_entry {
        ^~~~~~~~~~~~~~~~~~
/home/r...t/zephyr/include/shell/shell.h:108:14: note: and the implicitly-defined constructor does not initialize ‘const char* shell_static_entry::syntax’
  const char *syntax; /*!< Command syntax strings. */
              ^~~~~~
/home/r...t/zephyr/include/shell/shell.h:170:54: error: storage size of ‘shell_sub_h’ isn’t known
  static const struct shell_static_entry shell_##name[];                 \
                                                      ^
../src/main.cpp:104:1: note: in expansion of macro ‘SHELL_CREATE_STATIC_SUBCMD_SET’
 SHELL_CREATE_STATIC_SUBCMD_SET(sub_h)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/r...t/zephyr/include/shell/shell.h:172:12: error: expected primary-expression before ‘.’ token
            .u.entry =                \
            ^
../src/main.cpp:104:1: note: in expansion of macro ‘SHELL_CREATE_STATIC_SUBCMD_SET’
 SHELL_CREATE_STATIC_SUBCMD_SET(sub_h)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/r...t/zephyr/include/shell/shell.h:82:2: error: taking address of temporary [-fpermissive]
  }
  ^
/home/r...t/zephyr/include/shell/shell.h:213:52: note: in expansion of macro ‘SHELL_ARG’
   .args = _mandatory ? (&(struct shell_static_args)SHELL_ARG(    \
                                                    ^~~~~~~~~
/home/r...t/zephyr/include/shell/shell.h:227:2: note: in expansion of macro ‘SHELL_CMD_ARG’
  SHELL_CMD_ARG(_syntax, _subcmd, _help, _handler, 0, 0)
  ^~~~~~~~~~~~~
../src/main.cpp:106:5: note: in expansion of macro ‘SHELL_CMD’
     SHELL_CMD(t, NULL, "APP Power on self test status", shell_app_post),
     ^~~~~~~~~
../src/main.cpp:104:32: warning: ‘sub_h’ defined but not used [-Wunused-variable]
 SHELL_CREATE_STATIC_SUBCMD_SET(sub_h)
                                ^
/home/r...t/zephyr/include/shell/shell.h:171:38: note: in definition of macro ‘SHELL_CREATE_STATIC_SUBCMD_SET’
  static const struct shell_cmd_entry name = { .is_dynamic = false,      \
                                      ^~~~
cc1plus: warning: unrecognized command line option ‘-Wno-register’
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.1 LTS
 - Toolchain Zephyr SDK 0.9.5
 - Zephyr Version: zephyr-v1.13.0-4557-g2e68bc02b9
- git SHA: 2e68bc02b9a97cf24b3f74d92f3bc6444aa7749a

**Additional context**
The main issues I had to resolve was:

* _... error: uninitialized const ..._
* _... error: expected primary-expression before ‘.’ token .u.entry ..._
* _... error: taking address of temporary ..._
* _... note: ‘const struct shell_static_entry’ has no user-provided default constructor ..._

I could get the code running by change the initialization approach on the shell macros, but I had to change the way to do, so here is the current solution: 

```cpp
#include <misc/printk.h>
#include <shell/shell.h>
#include <shell/shell_fprintf.h>
#include <stdio.h>

/**
 * @brief Initializes a shell command arguments
 *
 * @param[in] _mandatory Number of mandatory arguments.
 * @param[in] _optional  Number of optional arguments.
 */
#define XXSHELL_ARG(_mandatory, _optional) \
    {                                      \
        _mandatory, _optional,             \
    }

#define XXSHELL_CMD_ARG(_syntax, _subcmd, _help, _handler, _mandatory, _optional)   \
    {                                                                               \
        (const char *) STRINGIFY(_syntax), (const char *) _help, _subcmd, _handler, \
        {                                                                           \
            _mandatory, _optional                                                   \
        }                                                                           \
    }

/**
 * @brief Macro for creating a subcommand set. It must be used outside of any
 * function body.
 *
 * @param[in] name	Name of the subcommand set.
 */
#define XXSHELL_CREATE_STATIC_SUBCMD_SET(name, ...)                                 \
    static const struct shell_static_entry shell_##name[] = {__VA_ARGS__,           \
                                                             SHELL_SUBCMD_SET_END}; \
    static const struct shell_cmd_entry name = {false, (shell_dynamic_get) shell_##name};

/**
 * @brief Initializes a shell command.
 *
 * @param[in] _syntax	Command syntax (for example: history).
 * @param[in] _subcmd	Pointer to a subcommands array.
 * @param[in] _help	Pointer to a command help string.
 * @param[in] _handler	Pointer to a function handler.
 */
#define XXSHELL_CMD(_syntax, _subcmd, _help, _handler)                                   \
    {                                                                                    \
        (const char *) STRINGIFY(_syntax), (const char *) _help, _subcmd, _handler, NULL \
    }

/**
 * @brief Macro for defining and adding a root command (level 0) with
 * arguments.
 *
 * @note All root commands must have different name.
 *
 * @param[in] syntax	Command syntax (for example: history).
 * @param[in] subcmd	Pointer to a subcommands array.
 * @param[in] help	Pointer to a command help string.
 * @param[in] handler	Pointer to a function handler.
 */
#define XXSHELL_CMD_REGISTER(syntax, subcmd, help, handler)                         \
    static const struct shell_static_entry UTIL_CAT(_shell_, syntax) =              \
        XXSHELL_CMD(syntax, subcmd, help, handler);                                 \
    static const struct shell_cmd_entry UTIL_CAT(shell_cmd_, syntax) __attribute__( \
        (section("." STRINGIFY(UTIL_CAT(shell_root_cmd_, syntax)))))                \
        __attribute__((used)) = {false, (shell_dynamic_get) &UTIL_CAT(_shell_, syntax)}

int shell_app_post(const struct shell *shell, size_t argc, char **argv)
{
    shell_print(shell, "post");
    return 0;
}
int shell_app_reboot(const struct shell *shell, size_t argc, char **argv)
{
    shell_print(shell, "reboot");
    return 0;
}

XXSHELL_CREATE_STATIC_SUBCMD_SET(sub_h,
                                 XXSHELL_CMD(t, NULL, "APP Power on self test status",
                                             shell_app_post));

int main(void)
{
    XXSHELL_CMD_REGISTER(h, &sub_h, "Info commands", NULL);
    XXSHELL_CMD_REGISTER(r, NULL, "General reset commands", shell_app_reboot);

    while (1) {
        k_sleep(K_SECONDS(10));
        printk("ping\n");
    }

    return 0;
}
```

jarz-nordic, 2019-02-11:
It is not possible to compile shell with C++. It is because C++ does not allow to declare a const variable and assign it later. I do not have a fix for that now and I have no idea when it can be done.

rodrigopex, 2019-02-11:
> It is not possible to compile shell with C++. It is because C++ does not allow to declare a const variable and assign it later. I do not have a fix for that now and I have no idea when it can be done.

The code I have done is working. Could you consider that as a solution? ;)

jarz-nordic, 2019-02-11:
ahh sorry! :) sure I will!

jarz-nordic, 2019-02-11:
@rodrigopex : It it shell API change. I have nothing agains it but it would be good to have it in Zephyr 1.14 because it is planned to be LTS. This requires lot of changes in already implemented commands. Do we have a volunteer? :)

@nashif @carlescufi : what do you think about it?

rodrigopex, 2019-02-11:
It would be great to help. Do you think the way that I have made the XX SHELL_CREATE_STATIC_SUBCMD_SET variant a good option? I think the biggest changes would be at the samples, right?

carlescufi, 2019-02-11:
@rodrigopex @jarz-nordic I don't think it would be an issue including this for 1.14, since ultimately the new shell broke C++ support and this would bring it back, so I would consider this a bugfix.

rodrigopex, 2019-02-11:
@carlescufi Ok. @jarz-nordic Should I start to write the code? I can help. I think the other macro would not change at all, so the "only" part that will need to be changed is the code related to SHELL_CREATE_STATIC_SUBCMD_SET.

marc-hb, 2019-02-11:
I think an API *change* would be easier to see in a commit (doesn't have to be PR yet). My 2 cents.

rodrigopex, 2019-02-11:
> I think an API _change_ would be easier to see in a commit (doesn't have to be PR yet). My 2 cents.

I will do that.

jarz-nordic, 2019-02-11:
@rodrigopex : As you noticed main problem is with `SHELL_CREATE_STATIC_SUBCMD_SET` and it is good idea to change it. Regarding updating all already existing commands we were thinking about some kind of script. It should not be a rocket science. I hope to have some proposal by end of tomorrow for that.

@rodrigopex thank you for this change!

rodrigopex, 2019-02-11:
@jarz-nordic I was trying to make it work inside the shell code, and I am facing issues with the dynamic shell when I change the code of the MACROs. It seems to be easy to change for static, but not so easy to dynamic. 

 This will not work in C++:
```cpp
#define SHELL_CMD_ARG(_syntax, _subcmd, _help, _handler,		      \
		      _mandatory, _optional) {				      \
	.syntax = (const char *)STRINGIFY(_syntax),			      \
	.subcmd = _subcmd,						      \
	.help  = (const char *)_help,					      \
	.handler = _handler,						      \
	.args = _mandatory ?						      \
	(&(struct shell_static_args) SHELL_ARG(_mandatory, _optional)) : NULL \  //<-- this will not work in C++
}
```
Maybe using alloca we could alloc the bytes. But could not set that inline.

nordic-krch, 2019-02-12:
> (&(struct shell_static_args) SHELL_ARG(_mandatory, _optional)) : NULL \  //<-- this will not work in C++

@jarz-nordic, regarding the issue above. It could be fixed by changing:
`const struct shell_static_args *args;` in `struct shell_static_entry` to
`const struct shell_static_args args;` and using values like `0xFF` to disable args checking.

jarz-nordic, 2019-02-12:
It it not a problem, we can document that.

jarz-nordic, 2019-02-12:
@rodrigopex : Please start a PR with your changes and add a macro: `XXSHELL_CREATE_STATIC_SUBCMD_SET`.

Next we will start another PR:
 1. Remove old `SHELL_CREATE_STATIC_SUBCMD_SET`
 2.  Rename your `XXSHELL_CREATE_STATIC_SUBCMD_SET` to `SHELL_CREATE_STATIC_SUBCMD_SET`
 3. Implement fix in `SHELL_ARG` proposed by @nordic-krch 
 4. Update all existing macros creating subcommands (we have prepared automated solution for that).

rodrigopex, 2019-02-12:
@jarz-nordic It will be necessary to modify other macros to make the XXSHELL_CREATE_STATIC_SUBCMD_SET work. It will be more productive if you add that, don't you think? I guess I will only interfere in a bad way here if I add that.  

@nordic-krch I was trying to change that yesterday. Is it possible to have the fields mandatory and optional both zero, right? I would guess it will be better to maintain that as the default value when using SHELL_CMD macros and set different values when using the SHELL_CMD_ARG.

jarz-nordic, 2019-02-12:
@rodrigopex : Ok we will add that and put you as a resolver in a commit message.

rodrigopex, 2019-02-12:
@jarz-nordic don't mind, the important is that it will run in C++. For help, I have done some code on my fork, you could take a look. It seems to not working properly yet, I am trying to figure out what is happening. https://github.com/rodrigopex/zephyr/tree/i13257

nashif, 2019-03-03:
Fixed by #13319

closed 2019-03-03 by nashif