zephyrproject-rtos/zephyr #13272: Catch all bug for build issues with SDK 0.10.0-beta2

galak, 2019-02-11:
Seeing the following build issues with SDK 0.10.0-beta, though some may not be limited to that SDK version.

```
15 tests selected, 85710 tests discarded due to filters
total complete:    0/  15   0%  failed:    0

qemu_xtensa               tests/lib/mem_alloc/libraries.libc.newlib          FAILED: build_error
	see: sanity-out/qemu_xtensa/tests/lib/mem_alloc/libraries.libc.newlib/build.log

total complete:    1/  15   6%  failed:    1

galileo                   samples/sensor/thermometer/test                    FAILED: build_error
	see: sanity-out/galileo/samples/sensor/thermometer/test/build.log

total complete:    2/  15  13%  failed:    2

altera_max10              tests/kernel/stack/stack_api/kernel.stack          FAILED: build_error
	see: sanity-out/altera_max10/tests/kernel/stack/stack_api/kernel.stack/build.log

total complete:    3/  15  20%  failed:    3

quark_d2000_crb           tests/drivers/spi/spi_loopback/periphera.spi       FAILED: build_error
	see: sanity-out/quark_d2000_crb/tests/drivers/spi/spi_loopback/periphera.spi/build.log

total complete:    4/  15  26%  failed:    4

quark_d2000_crb           tests/drivers/adc/adc_api/peripheral.adc           FAILED: build_error
	see: sanity-out/quark_d2000_crb/tests/drivers/adc/adc_api/peripheral.adc/build.log

total complete:    5/  15  33%  failed:    5

qemu_nios2                tests/kernel/stack/stack_api/kernel.stack          FAILED: build_error
	see: sanity-out/qemu_nios2/tests/kernel/stack/stack_api/kernel.stack/build.log

total complete:    6/  15  40%  failed:    6

mimxrt1050_evk_qspi       tests/drivers/spi/spi_loopback/periphera.spi       FAILED: build_error
	see: sanity-out/mimxrt1050_evk_qspi/tests/drivers/spi/spi_loopback/periphera.spi/build.log

total complete:    7/  15  46%  failed:    7

mimxrt1050_evk            tests/drivers/spi/spi_loopback/periphera.spi       FAILED: build_error
	see: sanity-out/mimxrt1050_evk/tests/drivers/spi/spi_loopback/periphera.spi/build.log

total complete:    8/  15  53%  failed:    8

mimxrt1050_evk_qspi       samples/cpp_synchronization/test                   FAILED: build_error
	see: sanity-out/mimxrt1050_evk_qspi/samples/cpp_synchronization/test/build.log

total complete:    9/  15  60%  failed:    9

qemu_xtensa               tests/subsys/jwt/libraries.encoding                FAILED: build_error
	see: sanity-out/qemu_xtensa/tests/subsys/jwt/libraries.encoding/build.log

total complete:   10/  15  66%  failed:   10

mimxrt1050_evk            samples/cpp_synchronization/test                   FAILED: build_error
	see: sanity-out/mimxrt1050_evk/samples/cpp_synchronization/test/build.log

total complete:   11/  15  73%  failed:   11

mimxrt1050_evk_qspi       tests/application_development/cpp/test             FAILED: build_error
	see: sanity-out/mimxrt1050_evk_qspi/tests/application_development/cpp/test/build.log

total complete:   12/  15  80%  failed:   12

mimxrt1050_evk            tests/application_development/cpp/test             FAILED: build_error
	see: sanity-out/mimxrt1050_evk/tests/application_development/cpp/test/build.log

total complete:   13/  15  86%  failed:   13

minnowboard               tests/kernel/mem_protect/mem_protect/kernel.memory_protection FAILED: build_error
	see: sanity-out/minnowboard/tests/kernel/mem_protect/mem_protect/kernel.memory_protection/build.log

total complete:   14/  15  93%  failed:   14

minnowboard               tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll FAILED: build_error
	see: sanity-out/minnowboard/tests/kernel/fifo/fifo_timeout/kernel.fifo.timeout.poll/build.log

total complete:   15/  15  100%  failed:   15
0 of 15 tests passed with 0 warnings in 10 seconds
```

galak, 2019-02-14:
altera_max10              tests/kernel/stack/stack_api/kernel.stack          FAILED: build_error

```
/opt/zephyr-sdk-0.10.0-rc2/nios2-zephyr-elf/bin/../lib/gcc/nios2-zephyr-elf/8.2.0/../../../../nios2-zephyr-elf/bin/ld: ../app/libapp.a(test_stack_contexts.c.obj): in function `tstack_pop':
/home/galak/git/zephyr/tests/kernel/stack/stack_api/src/test_stack_contexts.c:31: warning: unable to reach (null) (at 0x004002f4) from the global pointer (at 0x004082fc) because the offset (-32776) is out of the allowed range, -32678 to 32767

```

galak, 2019-02-14:
minowboard:

```
/opt/zephyr-sdk-0.10.0-rc2/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.2.0/../../../../i586-zephyr-elf/bin/ld:linker_pass_final.cmd:64 cannot move location counter backwards (from 000000000010afaf to 000000000010afae)
```

nashif, 2019-02-14:
@andrewboie you have an idea what the issue is with minnowboard and new toolchain?

nashif, 2019-02-20:
on minnowboard:

```
[100%] Linking C executable zephyr.elf
/home/nashif/Work/SDK/0.10.0-rc2-pr-43/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.2.0/../../../../i586-zephyr-elf/bin/ld:linker_pass_final.cmd:64 cannot move location counter backwards (from 000000000010be4e to 000000000010be4d)
```

nashif, 2019-02-20:
on nios2:

```
[ 94%] Linking C executable zephyr_prebuilt.elf
/home/nashif/Work/SDK/0.10.0-rc2-pr-43/nios2-zephyr-elf/bin/../lib/gcc/nios2-zephyr-elf/8.2.0/../../../../nios2-zephyr-elf/bin/ld: ../app/libapp.a(test_stack_contexts.c.obj): in function `tstack_pop':
/home/nashif/Work/zephyrproject/zephyr/tests/kernel/stack/stack_api/src/test_stack_contexts.c:31: warning: unable to reach (null) (at 0x00400300) from the global pointer (at 0x00408308) because the offset (-32776) is out of the allowed range, -32678 to 32767
```

nashif, 2019-02-20:
just noticed @galak already provided the trace :)

MaureenHelm, 2019-02-22:
@galak which issues are still outstanding? the nxp issues are resolved, so can we reassign this bug to the appropriate person?

nashif, 2019-02-22:
closing this now in favor of dedicated issue for the nios2 issue, I think I have a fix

closed 2019-02-22 by nashif