zephyrproject-rtos/zephyr #13276: Do we need to update fatfs

galak, 2019-02-11:
We seem to be using version 0.12b, there is now version 0.13c + patches at:

http://elm-chan.org/fsw/ff/00index_e.html

Does this need to be updated for 1.14 / LTS release?

galak, 2019-02-11:
@nashif you mentioned in the 1.14 release checklist about updating external modules, thus my query on this.

https://github.com/zephyrproject-rtos/zephyr/issues/9898

nvlsianpu, 2019-02-14:
I look into change log - Most of bugfixes are for exFat support which I believe is disabled in the zephyr.
Unless exFAT support is disabled i see a few fixes since 12.b:
-  Fixed successive f_getfree() can return wrong count on the FAT12/16 volume. (appeared at R0.12)
- Fixed configuration option _VOLUMES cannot be set 10. (appeared at R0.10c)
- Fixed f_fdisk() and f_mkfs() create the partition table with wrong CHS parameters. (appeared at R0.09) 
- Fixed f_setlabel() does not reject some invalid characters. (appeared at R0.09b)
- Fixed f_getcwd() cause output buffer overrun when the buffer has a valid drive number. (appeared at R0.13b)
- Fixed reading a directory gets infinite loop when the last directory entry is not empty. (appeared at R0.12)

nashif, 2019-02-19:
Changes since 0.12b

http://elm-chan.org/fsw/ff/updates.txt

```
R0.13c (October 14, 2018)
  Supported stdint.h for C99 and later. (integer.h was included in ff.h)
  Fixed reading a directory gets infinite loop when the last directory entry is not empty. (appeared at R0.12)
  Fixed creating a sub-directory in the fragmented sub-directory on the exFAT volume collapses FAT chain of the parent directory. (appeared at R0.12)
  Fixed f_getcwd() cause output buffer overrun when the buffer has a valid drive number. (appeared at R0.13b)

R0.13b (April 07, 2018)
  Added support for UTF-32 encoding on the API. (FF_LFN_UNICODE = 3)
  Added support for Unix style volume prefix. (FF_STR_VOLUME_ID = 2)
  Fixed accesing any object on the exFAT root directory beyond the cluster boundary can fail. (appeared at R0.12c)
  Fixed f_setlabel() does not reject some invalid characters. (appeared at R0.09b)

R0.13a (October 14, 2017)
  Added support for UTF-8 encoding on the API. (FF_LFN_UNICODE = 2)
  Added options for file name output buffer. (FF_LFN_BUF, FF_SFN_BUF).
  Added dynamic memory allocation option for working buffer of f_mkfs() and f_fdisk().
  Fixed f_fdisk() and f_mkfs() create the partition table with wrong CHS parameters. (appeared at R0.09)
  Fixed f_unlink() can cause lost clusters at fragmented file on the exFAT volume. (appeared at R0.12c)
  Fixed f_setlabel() rejects some valid characters for exFAT volume. (appeared at R0.12)

R0.13 (May 21, 2017)
  Changed heading character of configuration keywords "_" to "FF_".
  Removed ASCII-only configuration, FF_CODE_PAGE = 1. Use FF_CODE_PAGE = 437 instead.
  Added f_setcp(), run-time code page configuration. (FF_CODE_PAGE = 0)
  Improved cluster allocation time on stretch a deep buried cluster chain.
  Improved processing time of f_mkdir() with large cluster size by using FF_USE_LFN = 3.
  Improved NoFatChain flag of the fragmented file to be set after it is truncated and got contiguous.
  Fixed archive attribute is left not set when a file on the exFAT volume is renamed. (appeared at R0.12)
  Fixed exFAT FAT entry can be collapsed when write or lseek operation to the existing file is done. (appeared at R0.12c)
  Fixed creating a file can fail when a new cluster allocation to the exFAT directory occures. (appeared at R0.12c)

R0.12c (March 04, 2017)
  Improved write throughput at the fragmented file on the exFAT volume.
  Made memory usage for exFAT be able to be reduced as decreasing _MAX_LFN.
  Fixed successive f_getfree() can return wrong count on the FAT12/16 volume. (appeared at R0.12)
  Fixed configuration option _VOLUMES cannot be set 10. (appeared at R0.10c)
```

nashif, 2019-02-19:
My recommendation is not to update for 1.14 and make it high priority for 1.15.

nvlsianpu, 2019-02-20:
^^I agree with your opinion.

ioannisg, 2019-06-28:
@nashif @nvlsianpu @galak is anyone looking into this?

nashif, 2019-08-20:
This is not a bug.

dleach02, 2019-12-10:
> My recommendation is not to update for 1.14 and make it high priority for 1.15.

Moving milestone to 2.2.

nashif, 2020-08-04:
closing, there is no need to update this, this issue seems like a useless placeholder.

closed 2020-08-04 by nashif