zephyrproject-rtos/zephyr #13287: Zephyr can no longer apply DT overlays on a per-SoC basis

mbolivar, 2019-02-12:
When Kconfig ran before DTC, we could `ifdef #CONFIG_SOME_SOC` in our application DT overlays, and apply generic configuration on a per-SoC basis.

Now that DTC runs first, this is impossible. A solution was attempted in https://github.com/zephyrproject-rtos/zephyr/pull/12428 but apparently has been abandoned for LTS at the moment.

This regression makes it much more difficult to write portable applications that need to do something on a per-SoC basis, as Zephyr now only supports bringing in DT overlays on a per-board basis. Since many boards share SoCs, this hurts application portability.

One use case for SoC-level overlays is to place extra partitions not served by the default upstream per-board overlays, or override them entirely to suit application requirements.

A mechanism should be provided to change the device tree overlay depending on the SoC, not just the board.

galak, 2019-02-12:
Adding reference to PR #12073 as well.

galak, 2019-02-12:
The current proposal is to extract the compatibles from the soc node and convert them to `#defines`.  We'd ignore 'simple-bus'.

So for:
```
        soc {
                compatible = "silabs,efm32wg990f256", "silabs,efm32wg", "silabs,efm32", "simple-bus";
```

We'd have:

`-DDT_SOC_SILABS_EFM32WG990F256 -DDT_SOC_SILABS_EFM32WG -DDT_SOC_SILABS_EFM32`

galak, 2019-02-13:
@mbolivar I've been thinking about this some, and am not very keen and adding support into Zephyr for this by default.  I'm happy to help work on a script that can accomplish your goals and we provide in a contrib/ dir or something.  So you can run the script before you do a build to get the dts processed the way you need.

galak, 2019-03-07:
Closing this as I don't see this as a feature we want to generally support.  We can come up with some means of doing this as mentioned.

closed 2019-03-07 by galak