zephyrproject-rtos/zephyr #13289: tests/kernel/fifo/fifo_timeout fails on nrf52840_pca10056

arun1joshi, 2019-02-12:
The subtest _test_timeout_threads_pend_on_dual_fifos_ in tests/kernel/fifo/fifo_timeout fail with Assertion failure on nrf52840_pca10056 
Note: if I comment out this test case, there is no kernel panic and all other test cases pass in the suite; this test suite is passing on nrf52_pca10040

The test started failing from this commit: [[5fb4ac9](https://github.com/zephyrproject-rtos/zephyr/commit/5fb4ac94fb50a17e512f7544d423373eb362e122)]

**Arch**: arm
**Platforms**: nrf52840_pca10056

**Error on console log**:
````
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS zephyr-v1.13.0-4573-g012d404125 (delayed boot 1000ms) *****
Running test suite test_fifo_timeout
===================================================================
starting test - test_timeout_threads_pend_on_dual_fifos
 thread (q order: 0, t/o: 0, fifo 0x20000134)
 thread (q order: 5, t/o: 10, fifo 0x20000120)
 thread (q order: 1, t/o: 30, fifo 0x20000120)
 thread (q order: 7, t/o: 20, fifo 0x20000120)
 thread (q order: 2, t/o: 50, fifo 0x20000120)
 thread (q order: 8, t/o: 40, fifo 0x20000134)

    Assertion failed at zephyr.git/tests/kernel/fifo/fifo_timeout/src/main.c:130: test_thread_pend_and_timeout: is_timeout_in_range(start_time, d->timeout) is false

 thread (q order: 4, t/o: 70, fifo 0x20000134)
 thread (q order: 3, t/o: 80, fifo 0x20000134)
FAIL - test_timeout_threads_pend_on_dual_fifos
===================================================================
starting test - test_timeout_threads_pend_fail_on_fifo
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x4553534f
***** Hardware exception *****
Current thread ID = 0x20000c68
Faulting instruction address = 0x4a3c
Fatal fault in thread 0x20000c68! Aborting.
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x4553534f
***** Hardware exception *****
Current thread ID = 0x20000210
Faulting instruction address = 0x7482
Fatal fault in thread 0x20000210! Aborting.
ASSERTION FAIL [z_spin_lock_valid(l)] @ zephyr.git/include/spinlock.h:66
	Recursive spinlock
***** Kernel Panic! *****
Current thread ID = 0x20000294
Faulting instruction address = 0x1cba
evaluation failed
clean passed
COMPLETION failed

ioannisg, 2019-02-13:
@lemrey FYI

lemrey, 2019-02-13:
Oh, this is exactly what I have been hitting, though somewhere else entirely.
I would say this is a high priority bug.

lemrey, 2019-02-13:
I suspect this is a kernel bug, or some bug that was uncovered by recent kernel work at least. I am able to reproduce this assert consistently by running a modified `hello_world` sample, in [this](https://github.com/lemrey/zephyr/tree/bug-workq) tree.

This is far from my expertise and I might grossly off, but what I observe in that sample is that the first `k_delayed_work_submit` call executes its work handler before it has returned, thus spinlocking twice.

Perhaps @andyross could take a look.

lemrey, 2019-02-13:
A rudimentary augmentation of `k_spin_lock` and `k_delayed_work_submit_to_queue`  shows:

`Hello World! qemu_x86`
`k_delayed_work_submit_to_queue enters`
`>> spinlocking: 0x00400178`
`queue_insert`
`>> spinlocking: 0x00400128`
`_find_first_thread_to_unpend`
`>> spinlocking: 0x004000c4`
`<< spinrelease: 0x004000c4`
`_unpend_thread_no_timeout`
`>> spinlocking: 0x004000c4`
`<< spinrelease: 0x004000c4`
`_abort_timeout`
`>> spinlocking: 0x00400188`
`<< spinrelease: 0x00400188`
`_add_thread_to_ready_q`
`>> spinlocking: 0x004000c4`
`z_clock_elapsed`
`>> spinlocking: 0x00400030`
`<< spinrelease: 0x00400030`
`z_set_timeout_expiry`
`>> spinlocking: 0x00400188`
`z_clock_set_timeout`
`>> spinlocking: 0x00400030`
`<< spinrelease: 0x00400030`
`<< spinrelease: 0x00400188`
`<< spinrelease: 0x004000c4`
`k_delayed_work_submit_to_queue enters`
`>> spinlocking: 0x00400178`
`ASSERTION FAIL [z_spin_lock_valid(l)] @ ../../../../include/spinlock.h:72`
`        Recursive spinlock`

`k_delayed_work_submit_to_queue` was supposed to print something before returning; that doesn't happen.

andrewboie, 2019-02-13:
> it is reproducible on x86 and ARM.

Can you post a direct link to your test case which demonstrates this?

andyross, 2019-02-13:
Is that bisection accurate above?  I can totally imagine this being exposed by the spinlock validation patch or the later spinlockification series that went in last Friday.  I have a harder time seeing how double precision floating point is involved. :)

Can you re-bisect and check that it worked as of 552a03b48f825666d745275b94300963ee067c70?

andyross, 2019-02-13:
Actually reading that more carefully: it seems like the original root cause is the timing related failure reported via the test.  The recursive spinlock (which is a new assert) is more likely due to a bug in the fault handling.

lemrey, 2019-02-13:
> > it is reproducible on x86 and ARM.
> 
> Can you post a direct link to your test case which demonstrates this?

Hi, like I said it is reproducible in the `hello_world` sample in [this tree](https://github.com/lemrey/zephyr/tree/bug-workq), see commit: https://github.com/lemrey/zephyr/commit/4ad24dee172db966c532e959192493a7dab8844b

carlescufi, 2019-02-13:
@andyross we are seeing the same assertion on a downstream application that never used to fail before, so we suspect a regression somewhere.

carlescufi, 2019-02-14:
@andyross @andrewboie the root cause cannot be the timing related failure because @lemrey has reproduced the issue outside the scope of that test in the commit he links to [here](https://github.com/zephyrproject-rtos/zephyr/issues/13289#issuecomment-463324208)

ioannisg, 2019-02-14:
@andyross @lemrey  and all: I can also reproduce this with the modified Hello-World.
Also it is not a stack-overflow (Enabled HW_STACK_PROTECTION) and go no MPU errors.

carlescufi, 2019-02-14:
@ioannisg @lemrey have  you  bisected with @lemrey 's tree (
https://github.com/lemrey/zephyr/commit/4ad24dee172db966c532e959192493a7dab8844b)

lemrey, 2019-02-14:
a37a981b21d5e436e92c6160b5ad70e2ffd956ba is the first bad commit

    kernel/work_q: Spinlockify
    
    Each work_q object gets a separate spinlock to synchronize access
    instead of the global lock.  Note that there was a recursive lock
    condition in k_delayed_work_cancel(), so that's been split out into an
    internal unlocked version and the API entry point that wraps it with a
    lock.

@carlescufi @andyross

lemrey, 2019-02-19:
Neither I or @ioannisg have been able to reproduce this locally.
Anyone able to reproduce this?

carlescufi, 2019-02-20:
I cannot reproduce this on current master, neither with `qemu_cortex_m3` nor with `nrf52840_pca10056`

carlescufi, 2019-02-20:
Cleaned up the comments that were clobbering the history with a similar issue that is already covered and fixed.

andyross, 2019-02-26:
This reproduces for me on hardware (but not qemu).  And indeed it's in the spinlockification series.  It bisects down to commit 8a3d57b6cc60 (the userspace.c spinlockification).  Reverting just that one patch, or of course disabling userspace, suppresses the issue.

That's probably the most complicated of the spinlock changes, so this may take a bit of thinking.  But at least there's a path to release if I can't figure it out in time.

andyross, 2019-02-28:
So... funny story.

I reverted the userspace spinlockification patch that git bisect pointed me to, and slowly added it back until I found two spots that if changed from irq_lock to spinlock would trigger this bug.  But they didn't look so weird.  I then discovered that purely adding a local spin lock/unlock pair, with nothing inside it, would also trigger the bug, so I started worrying that the bug was in spinlocks themselves.  So I started stuffing all kinds of instrumentation in to try to track down why spinlocks weren't locking, but it all looks OK (yet didn't fix the bug).  Finally I investigated whether or not the timer driver was somehow interrupting spinlocks, but no, it's fine too.  Eventually I realized that disabling the spinlock validation layer entirely fixed the bug.  So maybe the problem was there?  But no.

So finally I got around to actually reading the test code.  And it was a tick aliasing issue in the test.  Sigh.

nRF sees these more easily due to the slow clock.  And spinlock validation when combined with userspace simply slowed down the test enough to put the aliased tick boundary into the right window.

PR #13970 (2019-03-01): tests/kernel/fifo/fifo_timeout: Tick alignment for oversensitive timing
This test was written to wait on a fifo with a timeout, return, and
check the timing between the start and end using k_cycle_get_32() to
see that it didn't run long.  But timeouts expire on tick boundaries,
and so if tick expires between the start of the test and the entry to
k_fifo_get(), the timeout will take one full tick longer than expected
due to aliasing.

As it happened this passed everywhere except nRF (whose cycle timer is
32 kHz and thus more susceptible to coarser aliasing like this), and
even there it passed f [... cut]
commit f085928a (2019-03-01): tests/kernel/fifo/fifo_timeout: Tick alignment for oversensitive timing

This test was written to wait on a fifo with a timeout, return, and
check the timing between the start and end using k_cycle_get_32() to
see that it didn't run long.  But timeouts expire on tick boundaries,
and so if tick expires between the start of the test and the entry to
k_fifo_get(), the timeout will take one full tick longer than expected
due to aliasing.

As it happened this passed everywhere except nRF (whose cycle timer is
32 kHz and thus more susceptible to coarser aliasing like this), and
even there it passed  [... cut]
closed 2019-03-01 by carlescufi