zephyrproject-rtos/zephyr #13291: samples/drivers/watchdog: Fatal fault in ISR

cinlyooi-intel, 2019-02-12:
**Describe the bug**
When attempting to feed the watch dog 5 times, a Fatal fault in ISR developed.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -GNinja -DBOARD=esp32 -DARCH=xtensa 
3. ninja
4. ninja run

**Screenshots or console output**
```
***** Booting Zephyr OS v1.14.0-rc1-8-gf81346cbd9 (delayed boot 1000ms) *****
Watchdog sample application
Feeding watchdog 5 times
Feeding watchdog...
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 3 PS 0x00060020 PC 0x4008188c VADDR 0x400805a4
 **  A0 0x800818fe  SP 0x3ffb11f0  A2 0x40080598  A3 0x00000000
 **  A4 0x00000001  A5 0x00060e23  A6 0x000003e7  A7 0x00000000
 **  A8 0x80081879  A9 0x3ffb11c0 A10 0x00000000 A11 0x00000000
 ** A12 0x00000000 A13 0x014f8b59 A14 0x00008386 A15 0x00061a80
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb0b00
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Feeding watchdog...
Waiting for reset...
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffb0000,len:144
load:0x3ffb0090,len:20
ho 0 tail 12 room 4
load:0x3ffb00a4,len:112
load:0x3ffb0114,len:2536
load:0x3ffb0e40,len:0
ho 12 tail 0 room 4
load:0x40080000,len:1024
load:0x40080400,len:256
ho 0 tail 12 room 4
load:0x40080500,len:144
load:0x40080590,len:8
load:0x40080598,len:16
ho 0 tail 12 room 4
load:0x400805a8,len:18420
load:0x400c0000,len:0
entry 0x40080974
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.14.0-rc1-8-gf81346cbd9 (delayed boot 1000ms) *****
Watchdog sample application
Feeding watchdog 5 times
Feeding watchdog...
 ** FATAL EXCEPTION
 ** CPU 0 EXCCAUSE 3 PS 0x00060020 PC 0x4008188c VADDR 0x400805a4
 **  A0 0x800818fe  SP 0x3ffb11f0  A2 0x40080598  A3 0x00000000
 **  A4 0x00000001  A5 0x00060e23  A6 0x000003e7  A7 0x00000000
 **  A8 0x80081879  A9 0x3ffb11c0 A10 0x00000000 A11 0x00000000
 ** A12 0x00000000 A13 0x014f8b59 A14 0x000090e2 A15 0x00061a80
 ** LBEG 0x00000000 LEND 0x00000000 LCOUNT 0x00000000
 ** SAR 0x00000000
Current thread ID = 0x3ffb0b00
Faulting instruction address = 0xdeaddead
Fatal fault in ISR! Spinning...
```
**Environment (please complete the following information):**
 - OS: linux
 - Toolchain: espressif
 - Commit SHA or Version used: f81346cbd90995a392c2994bddb2ad457dff26b8

SavinayDharmappa, 2019-03-06:
The commit 4844afa  enable CONFIG_LOG=y  for sample/drivers/watchdog which is causing the execption.If CONFIG_LOG is disabled the exception is not observed

SavinayDharmappa, 2019-03-11:
#14252

ManojSubbarao, 2019-03-26:
Duplicate of 10915. Closing the issue.

closed 2019-03-26 by ManojSubbarao