zephyrproject-rtos/zephyr #13300: NET: USB Ethernet tests were removed allowing to submit not compiling code

finikorg, 2019-02-12:
We had special overlays and configuration for USB Ethernet networking and now they are removed. Please restore.

tbursztyka, 2019-02-12:
Those overlays were previously in samples/net/echo_(server/client), restoring them into samples/net/sockets/echo_(server/client) would solve this

pfalcon, 2019-02-18:
> restoring them into samples/net/sockets/echo_(server/client) would solve this

Note that samples/net/sockets/echo_(server/client) themselves have some issues on real hardware so far. Ideally, there would be a discussion which sample app to use as a default "testing app" now (echo_server was such previously, YMMV). I for one would suggest  samples/net/sockets/echo (which also runs on Linux).

jukkar, 2019-02-18:
> I for one would suggest samples/net/sockets/echo (which also runs on Linux).

Hmm, it looks like a duplicate of echo-server so we could actually remove echo app in order to avoid confusion for the end user. Only thing extra it provides, is the ability to run in Linux. Not sure how useful thing that is as the sample apps are meant for zephyr.

pfalcon, 2019-02-18:
> Hmm, it looks like a duplicate of echo-server so we could actually remove echo app in order to avoid confusion for the end user.

No! Didn't we make any lessons from all these removals?

> Only thing extra it provides, is the ability to run in Linux.

And that's the tangible proof that it is a POSIX/BSD Sockets application, instead of mix of some adhoc APIs.

> Not sure how useful thing that is as the sample apps are meant for zephyr.

If everything goes right with Zephyr, 10 times more people will use it as a lightweight POSIXish RTOS to port the existing code, than write apps purposely for its own APIs.

But anyway, remembering what weird configs we had for net_app echo_server, they better go to socket `echo_server`, than `echo` indeed ;-).

PR #13485 (2019-02-18): samples: net: sockets: Add smoke testing support
The net-app based echo-server/client applications were used for
various smoke testing purposes. As those two apps are now removed
we need to introduce the same support into sockets based echo-client
and echo-server applications.

Fixes #13300 #13481
commit f626c3cc (2019-02-18): samples: net: sockets: Add smoke testing support

The net-app based echo-server/client applications were used for
various smoke testing purposes. As those two apps are now removed
we need to introduce the same support into sockets based echo-client
and echo-server applications.

Fixes #13300 #13481
closed 2019-02-18 by nashif