zephyrproject-rtos/zephyr #13306: Checking if UARTE TX complete on nRF52

tautologyclub, 2019-02-12:
**Describe the bug**
Function `uart_irq_tx_complete` does not work as intended. I want to use it to check if it's safe to turn off power to the device connected to UART (an RS485 transceiver in this case), which I can only do if I'm sure all bytes has been transmitted successfully. But `uart_irq_tx_complete` doesn't seem to help me here. From `uart_nrfx_uarte.c`:

	.irq_tx_ready		= uarte_nrfx_irq_tx_ready_complete,
	.irq_tx_complete	= uarte_nrfx_irq_tx_ready_complete,

So `tx_complete` will actually only check if uart controller is ready to accept new bytes. To me, this doesn't seem to be equivalent. If I were to shut off power to the RS485 transceiver just because `tx_complete` returned TRUE, I am pretty convinced I would drop bytes.

jarz-nordic, 2019-02-12:
Sorry, did you actually faced this problem on Nordic with current implementation or you are just complaining?

lemrey, 2019-02-12:
Hi @tautologyclub , I was a bit surprised when I looked at that first :)
I had a look at the implementation though and I'm convinced it is correct; plus, inspecting `uarte_nrfx_irq_tx_disable` shows that transmission will halt only once all bytes are out.

jarz-nordic, 2019-02-12:
@tautologyclub : In case of Nordic chip this both functions means the same hence I've implemented it like that. Different implementation would be for example for uC Atmega8. From its spec:

> The USART Transmitter has two flags that indicate its state: USART Data Register Empty
> (UDRE) and Transmit Complete (TXC). Both flags can be used for generating interrupts.

So in this case being ready to put new data to transmit buffer and fact that data has been shifted out are two different events. It is not a case for Nordic chips.

tautologyclub, 2019-02-12:
Yeah my issue came from trying to busy-wait for `uart_irq_tx_complete` inside the isr itself, which hung my program, heh. So what's the take-away here though? What's the proper way to ensure the uart fifo has been flushed, from outside isr context?

tautologyclub, 2019-02-12:
I mean strictly speaking "What's the proper way to ensure the uart fifo has been flushed, from outside isr context?" isn't really a bug, so let's just close this -- I'm sure you have plenty of things on your schedule with new release.

closed 2019-02-12 by tautologyclub