zephyrproject-rtos/zephyr #13318: kernel shell commands break real time semanticcs

mubes, 2019-02-12:
**Describe the bug**
When 'kernel stacks' or 'kernel threads' commands are entered at the command prompt realtime performance is lost.

**To Reproduce**

1) Enable shell, kernel extensions and thread/stack info. Specifically, the following must be defined;
CONFIG_INIT_STACKS, CONFIG_THREAD_MONITOR, CONFIG_THREAD_STACK_INFO
2) Setup short period secondary task which is dependent on external interrupt (e.g. to toggle a pin in response to incoming pin change)
3) Compile (tested on STM32F429 build)
4) At the command prompt type 'kernel stacks' or 'kernel threads'
5) Observe that realtime response of other task is lost.

I think this is due to the k_spin_lock in kernel/thread.c::k_thread_foreach ... would it not be sufficient to just do a k_sched_lock here? Worst consequence would be incorrectly reported threads/stacks for the case that the printing task manipulated one of the, which it's not supposed to do anyway.

**Expected behavior**
Interrupt responding second task should continue to operate correctly.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)
Annoyance. Prevents use of kernel threads/kernel stacks commands on a realtime system.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
Linux, Zephyr SDC, 1.3.99, a4e950a276a9d39a5bbe6495d085f78203e13a6c

andrewboie, 2019-02-14:
Is this really medium priority? The shell is a debugging feature..

nordic-krch, 2019-02-15:
Issue is about `k_thread_foreach` which is used by the shell command.

https://github.com/zephyrproject-rtos/zephyr/blob/95e8a582b4fbbef7ebaceb2924e4ddab3a6b213f/kernel/thread.c#L59

@andrewboie @andyross can you look at it? If after this functions is called realtime response is lost then this is something very wrong.

andyross, 2019-03-04:
Yeah, taking a lock around an iteration is bad practice for sure.

The only thing that lock is protecting is the threads list itself, there's absolutely no reason for it to be held while the callback is invoked.  Except that the list iteration is hand-coded and probably isn't safe vs. removal during iteration.

Fixing this correctly should:

1. Use an slist and not a hand-coded list for ?@#!! sake
2. Iterate with SYS_SLIST_FOR_EACH_NODE_SAFE()
3. Lock with a spinlock only around the list operations (i.e. one big lock around the loop, and unlock/relock around the callback)

Alternatively:

+ Check that no current callbacks passed to k_thread_foreach() attempt to create or destroy threads, and if they don't, document that as  a limitation and simply remove the lock entirely.

gnuless, 2019-03-05:
Unfortunately replacing the _kernel.threads list with an slist will break OpenOCD compatibility.  Both _kernel.threads and the next_thread member used to link them are exported.

andyross, 2019-03-05:
Ew.  Though you could still use an slist and just play offset games with the pointer fields, I guess.  And thinking about it, the whole design is poorly constrained.  It's obviously entirely possible for an interrupt or other context to preempt the iterator here, and clearly that context is allowed to create or abort threads too.

We might just need to wont-fix this one, honestly.  @mubes, what's the use case involved here?  If it's just a shell, maybe we don't care (or maybe we could fix it at a higher level by e.g. making a fast copy of the list in the iteration and then using that in the slow shell output).  If you have a real time app that needs to iterate on a thread list, then we should talk designs.

andrewboie, 2019-03-05:
> We might just need to wont-fix this one, honestly

+1

mubes, 2019-03-05:
Hi Andy,

I don't have a use case for the iteration (indeed, its only lightly used 
elsewhere) but I do need realtime response and this behavior breaks it 
completely :-( I only found the problem because it totally breaks my 
system which needs 100uS deterministic cyclic response. We can't go 
locking out interrupts for extended periods of time while something 
arbitrary happens! As far as I can understand the architecture I think 
locking the scheduler in the iterator (so other tasks don't mangle the 
list) and just putting a big warning in there that you shouldn't mess 
with the tasklist in an iterator callback would probably be enough for 
the short term, wouldn't it?  In any case I don't see that it kills the 
system even if tasks are created/destroyed...I think that would be 
caught by the termination test of the iterator loop. Nothing nasty would 
happen beyond you getting an out of date or incomplete task list back.

Regards

DAVE

gnuless, 2019-03-06:
> Nothing nasty would happen beyond you getting an out of date or incomplete task list back.

Except that I think any reasonable client of k_thread_foreach() is going to inspect the tasks in that list, and to do so without racing would require locking the scheduler, no?  So I believe the code is correct, as it is, at least for its intended purpose (debugging).

It's true that there are ways to make this facility more generally applicable, but I believe that's more a feature request rather than a bug.  So I'm going to keep this open but re-tag it and we'll see what kind of API design we can come up with for 1.15 that meets your requirements, @mubes.

mubes, 2019-03-06:
On 06/03/2019 17:28, Charles E. Youse wrote:
> Nothing nasty would happen beyond you getting an out of date or
>     incomplete task list back.
>
> Except that I think any reasonable client of k_thread_foreach() is
> going to inspect the tasks in that [...]
Hi Folks,

Using this totally breaks the real time semantics of the OS any time 
someone uses k_thread_foreach. Using it for debug purposes (i.e. when 
the debugger is active) is fine because you've already broken the 
realtime behavior at the point you wake up the debugger. It is _not_ 
fine to bork the RT behavior during normal operation. At the very least 
the UI commands that use this should be marked as breaking the RTOS and 
for this release we should consider patching them out.

(I'm not overly familiar with the internals of Zephyr, but in most OSes 
there's a distinction between disallowing interrupts disabling the 
scheduler...I have no problem with the scheduler going AWOL for a while 
(i.e. scheduler disable), but not responding to interrupts (i.e. 
interrupts_disable) for an indeterminate amount of time is a major issue).

Regards

DAVE

peter-mitsis, 2022-03-08:
In Nov 2019, the routine 'k_thread_foreach_unlocked()' was added to address this situation.

See https://github.com/zephyrproject-rtos/zephyr/pull/20937 for additional information.

I am closing this as it appears to have been addressed.

closed 2022-03-08 by peter-mitsis