zephyrproject-rtos/zephyr #13328: dts: erase-block-size should be available by partition

erwango, 2019-02-12:
**Is your enhancement proposal related to a problem? Please describe.**
Current implementation of nvs only support fixed sizes for erase operation.
Some SoC (stm32f7 for instance) have several flash sectors with variable sizes (4 sectors of 16Kb, 1 sector of 64Kb and 3 sectors of 128Kb) and support only block size erase.
In that case we can't define a single erase_block_size dt property.

Though, according to https://github.com/zephyrproject-rtos/zephyr/blob/1876f9785a3cce9d1e8b229baf472e1c9631037e/doc/reference/storage/nvs/nvs.rst#L40, NVS is declared with NVS_SECTOR_SIZE  "a multiple of the flash erase page size"

**Describe the solution you'd like**
It should be possible to define a erase_block_size depending of a flash partition so we can define a storage_partition in a specific sector with a defined erase_block_size.

**Describe alternatives you've considered**
Support variable flash erase page size in NVS

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

Laczen, 2019-02-13:
Hi @erwango, you should use flash page layout to receive the page size. In PR12892 that improves the startup it is also checked that the defined sector size is ok.

erwango, 2019-02-13:
@Laczen , so concretely, how can I run `samples/subsys/nvs/` with for instance nucleo_f746zg ?
```
#define NVS_SECTOR_SIZE DT_FLASH_ERASE_BLOCK_SIZE /* Multiple of FLASH_PAGE_SIZE */
#define NVS_SECTOR_COUNT 3 /* At least 2 sectors */
#define NVS_STORAGE_OFFSET DT_FLASH_AREA_STORAGE_OFFSET /* Start address of the
							 * filesystem in flash
							 */
static struct nvs_fs fs = {
	.sector_size = NVS_SECTOR_SIZE,
	.sector_count = NVS_SECTOR_COUNT,
	.offset = NVS_STORAGE_OFFSET,
};
```
At least I should be able to make it compile.

Laczen, 2019-02-13:
@erwango, I could say "The nucleo_f746zg is not a supported platform for the sample." (it is not a supported platform exactly for this reason), but to make it compile:
a. Set the sizes directly in the struct:
``
static struct nvs_fs fs = {
    .sector_size=2048,
    .sector_count=3
    .offset = 0x2f000
}
``
b. If you really have no idea what the pagesize is for a specific location where you would like to put the nvs file system you can get the pagesize using the flash page layout api, you can fill in the sector_size just before calling nvs_init().

On systems with "variable" flash page sizes you should use the page layout api to get the correct flash page size at a specific location. Providing support for a nvs fs that overlaps regions with different page size is not supported at the moment.

erwango, 2019-02-13:
> "The nucleo_f746zg is not a supported platform for the sample." (it is not a supported platform exactly for this reason)

Hence my current request. It would be nice to support dts configuration for SoC with heterogeneous flash page sizes. It could be optional flags that could be set in storage_partition node if erase-block-size isn't available. Limitation of nvs limited to one region in that case is not an issue I think.

Laczen, 2019-02-13:
@erwango, hence my original reply. This is already provided by the flash page api:
`rc = flash_get_page_info_by_offs(fs->flash_device, fs->offset, &info);`
`info.size` contains the page_size, no need to generate it.

erwango, 2019-02-13:
@Laczen, then why can't we get the sample work that way? and why does it requires `erase-block-size` to be available?

Laczen, 2019-02-13:
@erwango, if you make an issue nvs sample should not use DT_FLASH_ERASE_BLOCK_SIZE, I will change the sample.
Why it is erase-block-size: I don't really know, in nrf51 the flash_page_size is what is erased whenever a erase is done, so I guess erase_block_size = page_size. Maybe this is different on your mcu.

erwango, 2019-02-13:
> @erwango, if you make an issue nvs sample should not use DT_FLASH_ERASE_BLOCK_SIZE, I will change the sample.

@Laczen , ok thanks will do.

> Maybe this is different on your mcu.

This is the same for most of the ones I have in mind (block erase). Though, in some series, block size may vary.

erwango, 2019-02-13:
Raised #13369

closed 2019-02-13 by erwango